djphazer/O_C-Phazerville #3: TM2 Trigger is affected by slew amount

nickstokes, 2023-03-11:
Loving TM2, but noticed that the trigger length has the slew value applied. At slew=0 this makes the pulse short enough that my Plaits trigger input doesn't reliably pick it up.

djphazer, 2023-03-12:
This is mostly intentional. I like to use it as a Decay envelope tail with higher Slew values... and I kind of like being able to have super short triggers as well, for ridiculously snappy/clicky behavior. A Slew value of 4 is a pretty good compromise.

I hear you though, it becomes an issue when you just want a regular trigger. Perhaps I should make a separate output type, but it's already pretty crowded!

I'll probably compromise and make it behave like a normal trigger pulse when Slew <= 4

djphazer, 2023-03-28:
Resolved in https://github.com/djphazer/O_C-BenisphereSuite/commit/73e44fe27927bcec1eddcedb210e2edffe6bbc6d
Available in #7

I'm still conflating Triggers with Decay envelopes, but I think it's a fun feature that works ok as either.

closed 2023-03-28 by djphazer