djphazer/O_C-Phazerville #5: SBD_MIDI start error; cant compile

synthi, 2023-03-13:
Trying to complile for my buchla 2OC, using Arduino IDE 1-8-1 + teensyduino.
It kkeps getting the following error:

`HEM_ClockSetup:48: error: 'class usb_midi_class' has no member named 'Start'
             usbMIDI.sendRealTime(usbMIDI.Start);

                                          ^

HEM_ClockSetup:52: error: 'class usb_midi_class' has no member named 'Stop'
             usbMIDI.sendRealTime(usbMIDI.Stop);

                                          ^

HEM_ClockSetup:54: error: 'class usb_midi_class' has no member named 'Clock'
         if (clock_m->IsRunning() && clock_m->MIDITock()) usbMIDI.sendRealTime(usbMIDI.Clock);

                                                                                       ^

C:\Users\Usuario\Documents\Arduino\O_C-BenisphereSuite-cal8app-beta\software\o_c_REV\ORIG_HEM_ClockSetup.ino: At global scope:

ORIG_HEM_ClockSetup:23: error: redefinition of 'class ClockSetup'
 class ClockSetup : public HemisphereApplet {

       ^

HEM_ClockSetup:21: error: previous definition of 'class ClockSetup'
 class ClockSetup : public HemisphereApplet {

       ^

ORIG_HEM_ClockSetup:252: error: redefinition of 'ClockSetup ClockSetup_instance [1]'
 ClockSetup ClockSetup_instance[1];

                                 ^

HEM_ClockSetup:239: error: 'ClockSetup ClockSetup_instance [1]' previously declared here
 ClockSetup ClockSetup_instance[1];

            ^

ORIG_HEM_ClockSetup: In function 'void ClockSetup_Start(bool)':
ORIG_HEM_ClockSetup:254: error: redefinition of 'void ClockSetup_Start(bool)'
 void ClockSetup_Start(bool hemisphere) {ClockSetup_instance[hemisphere].BaseStart(hemisphere);}

      ^

HEM_ClockSetup:241: error: 'void ClockSetup_Start(bool)' previously defined here
 void ClockSetup_Start(bool hemisphere) {ClockSetup_instance[hemisphere].BaseStart(hemisphere);}

      ^

ORIG_HEM_ClockSetup: In function 'void ClockSetup_Controller(bool, bool)':
ORIG_HEM_ClockSetup:256: error: redefinition of 'void ClockSetup_Controller(bool, bool)'
 void ClockSetup_Controller(bool hemisphere, bool forwarding) {ClockSetup_instance[hemisphere].Controller();}

      ^

HEM_ClockSetup:243: error: 'void ClockSetup_Controller(bool, bool)' previously defined here
 void ClockSetup_Controller(bool hemisphere, bool forwarding) {ClockSetup_instance[hemisphere].Controller();}

      ^

ORIG_HEM_ClockSetup: In function 'void ClockSetup_View(bool)':
ORIG_HEM_ClockSetup:257: error: redefinition of 'void ClockSetup_View(bool)'
 void ClockSetup_View(bool hemisphere) {ClockSetup_instance[hemisphere].BaseView();}

      ^

HEM_ClockSetup:244: error: 'void ClockSetup_View(bool)' previously defined here
 void ClockSetup_View(bool hemisphere) {ClockSetup_instance[hemisphere].BaseView();}

      ^

ORIG_HEM_ClockSetup: In function 'void ClockSetup_OnButtonPress(bool)':
ORIG_HEM_ClockSetup:258: error: redefinition of 'void ClockSetup_OnButtonPress(bool)'
 void ClockSetup_OnButtonPress(bool hemisphere) {ClockSetup_instance[hemisphere].OnButtonPress();}

      ^

HEM_ClockSetup:245: error: 'void ClockSetup_OnButtonPress(bool)' previously defined here
 void ClockSetup_OnButtonPress(bool hemisphere) {ClockSetup_instance[hemisphere].OnButtonPress();}

      ^

ORIG_HEM_ClockSetup: In function 'void ClockSetup_OnEncoderMove(bool, int)':
ORIG_HEM_ClockSetup:259: error: redefinition of 'void ClockSetup_OnEncoderMove(bool, int)'
 void ClockSetup_OnEncoderMove(bool hemisphere, int direction) {ClockSetup_instance[hemisphere].OnEncoderMove(direction);}

      ^

HEM_ClockSetup:246: error: 'void ClockSetup_OnEncoderMove(bool, int)' previously defined here
 void ClockSetup_OnEncoderMove(bool hemisphere, int direction) {ClockSetup_instance[hemisphere].OnEncoderMove(direction);}

      ^

ORIG_HEM_ClockSetup: In function 'void ClockSetup_ToggleHelpScreen(bool)':
ORIG_HEM_ClockSetup:260: error: redefinition of 'void ClockSetup_ToggleHelpScreen(bool)'
 void ClockSetup_ToggleHelpScreen(bool hemisphere) {ClockSetup_instance[hemisphere].HelpScreen();}

      ^

HEM_ClockSetup:247: error: 'void ClockSetup_ToggleHelpScreen(bool)' previously defined here
 void ClockSetup_ToggleHelpScreen(bool hemisphere) {ClockSetup_instance[hemisphere].HelpScreen();}

      ^

ORIG_HEM_ClockSetup: In function 'uint64_t ClockSetup_OnDataRequest(bool)':
ORIG_HEM_ClockSetup:261: error: redefinition of 'uint64_t ClockSetup_OnDataRequest(bool)'
 uint64_t ClockSetup_OnDataRequest(bool hemisphere) {return ClockSetup_instance[hemisphere].OnDataRequest();}

          ^

HEM_ClockSetup:248: error: 'uint64_t ClockSetup_OnDataRequest(bool)' previously defined here
 uint64_t ClockSetup_OnDataRequest(bool hemisphere) {return ClockSetup_instance[hemisphere].OnDataRequest();}

          ^

ORIG_HEM_ClockSetup: In function 'void ClockSetup_OnDataReceive(bool, uint64_t)':
ORIG_HEM_ClockSetup:262: error: redefinition of 'void ClockSetup_OnDataReceive(bool, uint64_t)'
 void ClockSetup_OnDataReceive(bool hemisphere, uint64_t data) {ClockSetup_instance[hemisphere].OnDataReceive(data);}

      ^

HEM_ClockSetup:249: error: 'void ClockSetup_OnDataReceive(bool, uint64_t)' previously defined here
 void ClockSetup_OnDataReceive(bool hemisphere, uint64_t data) {ClockSetup_instance[hemisphere].OnDataReceive(data);}

      ^

'class usb_midi_class' has no member named 'Start'
`

djphazer, 2023-03-13:
Older versions of the teensyduino framework were missing definitions for `usbMIDI.Start`, `usbMIDI.Stop` and `usbMIDI.Clock`. I've abandoned support for the older Arduino IDE in favor of Platform IO, but you might still succeed if you revert the relevant lines from this commit: https://github.com/djphazer/O_C-BenisphereSuite/commit/f741b4d58f8735c20a158cedecaa89ee67214f9c

I honestly should have left it alone for better compatibility, but this serves as a gentle suggestion to upgrade your toolchain!

synthi, 2023-03-13:
Thanks! will try to revert vhanges... The problem seem to be that the Buchla builds only work when compiled with theese older tools... when compiling with updated versions the triggers don´t work... there is sone discussion here: https://modwiggler.com/forum/viewtopic.php?t=229917&start=25

synthi, 2023-03-13:
I managed to download platformIO and complile the firmware! And seem to be working for Buchla! Thanks!

djphazer, 2023-03-13:
Great news! This affirms that the newer build toolchain is indeed stable and robust. 🦾

Perhaps let the forum folks know of your success. Closing this issue as complete.

closed 2023-03-13 by djphazer