djphazer/O_C-Phazerville #6: EBBandLFO Buchla 1.2V/oct tracking

synthi, 2023-03-13:
Trying Ebband LFO, wonderfull! sounding great in audio rate as an oscillator, and VC of slope,etc... Would be perfect as an aditional oscillator for my Easel, I see it have a V/oct input I don´t know if could be moded for 1.2V/oct?? Thanks!

djphazer, 2023-03-13:
It's certainly possible! I'll see what I can do when I have time 😉

synthi, 2023-03-14:
ok! trying again and its already working!! it was a fault compiling the firmware without the buchla flags. What is not working is the bipolar output, in LFO mode only the upper part works, I think It need a kind of offset

djphazer, 2023-03-14:
So does EbbAndLFO track correctly with 1.2V/Oct input if the compiler flags are enabled?

For the bipolar output: Do other applets successfully output negative voltage? I don't know much about these Buchla variants, seems like it's only positive voltage...

djphazer, 2023-03-28:
This seems mostly resolved. Looks like the appropriate build flags will cause all inputs and outputs to be scaled appropriately. I'm still not sure what should be done about a bias offset for bipolar outputs... and I can't test this myself. Feedback from more Buchla users would be helpful.

I added some configuration to the `platformio.ini` file to enable building for Buchla, in case others want to try it out:
```
pio run -e buchla -t upload
```
I didn't put the `-DBUCHLA_cOC` option in there tho, you might need that.