djphazer/O_C-Phazerville #8: Feature Idea: Subharmonicon

yratof, 2023-03-29:
Been playing around with the code to try to make a subharmonicon applet, but i'm not experienced enough to get it running

I think it would be two apps

First app would:

1) Take in a frequency into the CV
2) Spit out 2 harmonic frequencies based on user selected division

Second app would take in a clock and advance a sequence, but you would set how long each of 4+ gates would stay open for.
1) Set clock division for each step

Here is how I'm picturing the UI:

[image]

How mad am I to suggest this?

yratof, 2023-03-29:
> Very cool idea! The sequencer part sounds like something I would immediately enjoy...
> To be clear, would the harmonic divider part deal with audio rate? or pitch CV?

Im assuming it would work like a quantiser, but dividing it by the subharmonic division

yratof, 2023-04-17:
On this - do we have any apps that can take a clock and click divide it, but each divider waits for previous divisions trigger before dividing

djphazer, 2023-04-17:
There's the standard **[Clock Div](https://github.com/Chysn/O_C-HemisphereSuite/wiki/Clock-Divider-Multiplier)** applet.
There's also Beni's stochastic [**ProbDiv**](https://github.com/benirose/O_C-BenisphereSuite/wiki/ProbDiv) applet.

Nothing quite like the Subharmonicon sequencer tho, this is a unique applet idea - a deliberate clock division sequencer.
I also need to figure out the math involved to convert a harmonic ratio into a V/Oct offset for the subharmonizer pitch applet.

yratof, 2023-04-17:
It's going to be a great tool

vestedaf, 2023-05-15:
This is an awesome idea! I need a second o_C so badly...

yratof, 2023-10-04:
Did this idea ever make it past this suggestion?

djphazer, 2023-10-04:
Unfortunately not yet...

djphazer, 2023-10-27:
Sorry for sleeping on this one... after some discussion on Discord about using the O_C for (sub)harmonic series quantization, I'm now aware of some scales already present in the firmware that might help realize this app idea. I may try to tackle it after the big weekend...

yratof, 2023-10-27:
Sounds like a fantastic halloween plan :D

yratof, 2023-12-18:
Ayup - How are plans going for anything such as the above ;)

yratof, 2024-01-31:
Closing this issue as we now have the [Polydiv](https://github.com/djphazer/O_C-Phazerville/blob/PSv1.7-beta1/software/src/applets/PolyDiv.h) & [DivSeq](https://github.com/djphazer/O_C-Phazerville/blob/PSv1.7-beta1/software/src/applets/EbbAndLfo.h) in 1.7.beta1 resolving the feature request

closed 2024-01-31 by yratof