djphazer/O_C-Phazerville #10: Add the Harmonizer fork based on DualQuantApp

yratof, 2023-04-12:
Found this today: https://www.youtube.com/watch?v=rp_AFcJDCRc

Uses this fork's HEM applet:
https://github.com/kfirmanty/O_C-HemisphereSuite/commit/2b053dadb6297383ab7e9828b5e69b3322a2c878

Definitely worth adding to this repo, if there is space available

djphazer, 2023-04-12:
Oh neat, I'll check this out! I have a hunch it's probably similar to Chordinator or Squanch...

djphazer, 2023-04-15:
I pulled this in here: https://github.com/djphazer/O_C-BenisphereSuite/commit/d0a26396c505798e931be59c4774746946dd9d55

I like the premise - it spits out 2 pitches above the input by scale degree. The UI could still use some beautification...

djphazer, 2023-04-17:
After playing with this a bit, I've concluded that it does, in fact, do exactly what **[Squanch](https://github.com/Chysn/O_C-HemisphereSuite/wiki/Squanch---Shifting-Quantizer)** already does.
Squanch also has more functionality: clocked operation, transpose down as well as up, CV2 input added to the 2nd output, octave jump via gate on TR2.

closed 2023-04-17 by djphazer

yratof, 2023-04-17:
Yeah, was hoping for something more - good to know that we’re under-utilising squanch

djphazer, 2023-04-17:
I also realized that Squanch lacks a Root Note parameter... so I'm adding it in right meow!