djphazer/O_C-Phazerville #13: Clock setup when pressing up + down in sequence

jtomasrl, 2023-05-23:
When pressing Up, releasing and immediately pressing down, the Clock setup menu is displayed (v1.5.1). Same happen if pressing Down first and then Up.

djphazer, 2023-05-23:
Correct. Is this undesirable to you?

The code detects double-clicks on the Select buttons and toggles Clock Setup when the 2nd click is not the same button as the 1st click. So the actual gesture is "press both buttons in quick succession" which also covers "simultaneously"

djphazer, 2023-05-23:
I like this because it means you can still get to the clock screen with one finger... when the cables are in the way 😉

I could potentially make the double-click time shorter so this is harder to stumble upon accidentally... which would also affect the applet help screen function.

jtomasrl, 2023-05-24:
When navigating through menus with +- navigation, it can get quite tricky. Maybe the Hemisphere config should be a menu entry  (in the apps menu) instead of overwriting its default behavior (clock setup)

jtomasrl, 2023-05-24:
I think this was an expected behavior. Anyways I made a change in my code, I'll leave it here is someone else wants to change the default behaviour:

`HemisphereApplet.h`, add

```c++
#define HEMISPHERE_SIM_CLICK_TIME 1000
``` 

Then, in `APP_HEMISPHERE.ino`, inside `DelegateSelectButtonPush` function, after `if (down)` section, change

```c++
            if (OC::CORE::ticks - click_tick < HEMISPHERE_DOUBLE_CLICK_TIME) {
                // This is a double-click. Activate corresponding help screen or Clock Setup
                if (hemisphere == first_click)
                    SetHelpScreen(hemisphere);
                else // up + down simultaneous
                    clock_setup = 1;

                // leave Select Mode, and reset the double-click timer
                select_mode = -1;
                click_tick = 0;
           }
``` 

With

```c++
            if (OC::CORE::ticks - click_tick < HEMISPHERE_SIM_CLICK_TIME) {
                clock_setup = 1;
                click_tick = 0;
            } else if (OC::CORE::ticks - click_tick < HEMISPHERE_DOUBLE_CLICK_TIME) {
                // This is a double-click. Activate corresponding help screen or Clock Setup
                if (hemisphere == first_click)
                    SetHelpScreen(hemisphere);
                    
                // leave Select Mode, and reset the double-click timer
                select_mode = -1;
                click_tick = 0;
            }
``` 

This will add a second Click timer, for simultaneous click, so it will simulate a simultaneous click on Up and Down

djphazer, 2023-05-26:
I actually quite like this solution! I'll try it out, see how it feels.

djphazer, 2023-05-31:
Fixed in v1.6.1
https://github.com/djphazer/O_C-BenisphereSuite/commit/f45dcb515e41777871059b09e3cb46b99132b548

closed 2023-05-31 by djphazer