djphazer/O_C-Phazerville #15: Knob directions reversed in flipped main build?

discohead, 2023-06-01:
Just tried out the flipped build for the first time and I guess the knobs are not "flipped", because turning right moves left/moves up/decreases and turning left moves rights/moves down/increases. Wasn't sure if this is a known issue or not, if so, feel free to close. I'm running v1.6.1

djphazer, 2023-06-01:
I actually considered this a standard calibration difference before now... you can adjust the encoders' direction on the Calibration screens in the "Setup / About" app. (press either Up/Down buttons to cycle through)

But I could probably invert that setting as well for the flipped build and save us all the trouble 😉 Thanks for the nudge

discohead, 2023-06-01:
Got it. Thanks for the tip. Yea, making `R reversed` the default setting for flipped builds seems like good idea..

closed 2023-06-01 by discohead

djphazer, 2023-06-16:
I pushed a fix for this here: https://github.com/djphazer/O_C-BenisphereSuite/commit/cb06389fe4106c3e5bbf7e150fe334ecfa4f1de9
The next release that includes it will require you to adjust the calibration once, but never again ;)