djphazer/O_C-Phazerville #16: Vector LFO issue

X-modular, 2023-06-15:
Vector LFO has been updated to include speeds below 0.10hz.  Thanks for that.  However, there is a bug.  The LFO speed doesn't save right.  It defaults to 4.92hz for both LFOs.  Other parameters such as shape save fine.

djphazer, 2023-06-16:
It sounds like they've been clocked at 74 BPM... about a 203 ms pulse.
Actually, I see this is a bigger issue that I've noticed - all digital inputs fire a clock on power up, which is a bit of a nuisance. Does it do this on other Hemisphere forks? Would be a nice bug to squash!

Save/load appears to work fine when switching presets.
I do think I'd like to rework the frequency setting tho. I feel like somebody already did that somewhere... made it V/Oct

X-modular, 2023-06-16:
The LFO speed would save fine in the original Hemispheres I was using.  The Vector LFO app had other bugs such as not being able to control the speed with CV.  And the minimum LFO speed was 0.10hz.  Someone updated the app since then.  Maybe you did?  The LFO speed can go down to 0.01hz now, which is great.  Not sure if the CV control of speed has been fixed.  

It would be nice to get the Vector apps working right.  They are the only modulation apps that can use user defined waveshapes.  Which brings me to my next problem.  The app that is used to create new waveshapes isn't part of Phaserville.

djphazer, 2023-06-18:
This was a bug with my manual clock triggers, fixed here: https://github.com/djphazer/O_C-BenisphereSuite/commit/ecc8f8717d41bdf47a4211fabec73638dce1d581

There have been a few modifications to the Vector LFO applet, the most recent being my rework of the blend behavior of output 2, controlled by CV2 - https://github.com/djphazer/O_C-BenisphereSuite/commits/phazerville/software/o_c_REV/HEM_VectorLFO.ino

CV1 input controls frequency. And the Waveform Editor is still in the App menu! I never removed it. I admittedly haven't tested it that much, but it appeared to work fine last night.

X-modular, 2023-06-19:
Ok, great.  Thanks.  The update to the Vector LFO app is appreciated too.  The original configuration of CV2 was less useful at the default setting.  

Sorry I missed the waveform editor.  Good to know its still there.  

Any idea when the update with this bug fix will be available for download?

djphazer, 2023-06-20:
End of the week, if not sooner!

X-modular, 2023-06-20:
Awesome.  Thanks very much for doing this.  I really appreciate it.

closed 2023-06-23 by djphazer