djphazer/O_C-Phazerville #17: CV 'bleed' in Hemispheres

kaffeetrinken, 2023-07-15:
I noticed some 'bleed' of CV in Hemispheres.
CV2 from right hemisphere influences erratically CV3 in right hemisphere (see pictures attached, CV2 changes clock division in right hemisphere).

[image]
[image]

djphazer, 2023-07-15:
Does this happen in other firmwares, too? It might be a hardware issue with your unit...

djphazer, 2023-07-15:
I was able to reproduce this briefly. I'll look into it further.

djphazer, 2023-07-20:
I don't have a solution yet, but I think I know what's going on... we're pushing the Teensy to the limit!
It seems like there are cases where one processing cycle will overrun the 60us timer and indeed bleed into the next cycle. CV input reads are staggered, spread across 4 cycles. So I'm seeing CV2 bleed into CV3, and also CV4 bleeds into CV1.

TB-3PO is one of the offending applets that occasionally takes too long; Game of Life will also do it. The only applets currently susceptible to the bleed are ClockDiv, ProbMeloD and ProbDiv - where a spontaneous CV input will change a value, but a 0V input will not. I'll most likely modify these applets to make the CV input a modulation of the current setting, rather than an absolute value, and that will mitigate the problem. To actually eliminate the bleed will require some careful optimizations or restructuring of core execution patterns...

djphazer, 2023-08-04:
Possible fix on the dev/1.6 branch: https://github.com/djphazer/O_C-BenisphereSuite/commit/d775695edbee1aeed9dca513cfbf0a5bc6963d7f

Switching to DMA-based ADC probably eliminates bleed, needs testing...

djphazer, 2023-08-15:
Just published v1.6.3 with the DMA patch that should fix this. Feel free to reopen if you can still reproduce.

closed 2023-08-15 by djphazer