djphazer/O_C-Phazerville #18: Quadraturia does not remember VOR settings

adamjmurray, 2023-07-28:
I'm running `o_C-phazerville-v1.6.2+VOR-2224121e.hex` on a a Plum Audio OCP X. When using the Quadraturia app, saving state will not remember the VOR settings. Long-pressing the right encoder to go to the app menu and then immediately returning to Quadraturia loses the VOR setting and it resets to -5/5V range, so the setting may already be lost before any attempt to save state can be made.

Not sure if this is an issue with other apps. It might even be a long-standing O_C issue. All I can say for sure is I'm using Quadraturia a lot lately and I have to remember to fix the VOR setting every time I power on my synth.

djphazer, 2023-08-03:
The Vbias setting is not currently stored in EEPROM for any app... it's a nice idea that was never implemented. I've thought about adding it as part of Hemisphere Config. I'll consider injecting it into the Quadraturia settings when I can.

There is an auto-config mechanism that sets certain ranges for each app when you switch to it - for Quadraturia, it's the +/-5V range.

Out of curiosity, which range are you using?

adamjmurray, 2023-08-03:
> The Vbias setting is not currently stored in EEPROM for any app... it's a nice idea that was never implemented.

Ah ok. That makes sense. I guess the majority of the time I am happy with the vbias auto-config behavior so it took me a while to notice this long-standing limitation.

It would be cool if it's supported one day, and I understand Hemispheres support is usually the priority.

> Out of curiosity, which range are you using?

I normally would use +/-5V, and I think that makes sense as the default for Quadraturia since it's a quad LFO and LFOs are normally bipolar.

I encountered this bug because I've recently become very interested in sampling and holding cyclical modulation instead of noise. I am using a ton of S&H modules, including Mutable Instruments Stages to do S&H. But Stages is unipolar and can only process signals in the range 0-10V. So, I switched my O_C VOR to 0-10V so I can feed the Quadraturia output into Stages. Probably a pretty unique use case. Although, there are plenty of situations someone might want unipolar positive modulation signals, so I'm sure other people will run into this.

djphazer, 2023-08-15:
Implemented in v1.6.3 - try it out!

closed 2023-08-15 by djphazer

adamjmurray, 2023-08-25:
Works great during my initial tests. Thanks!