djphazer/O_C-Phazerville #19: Error building

jtomasrl, 2023-08-09:
Im getting an error while building with a change in platformio.ini.

Specifically related to this line: 
extra_scripts = pre:resources/progname.py

https://github.com/djphazer/O_C-BenisphereSuite/blob/dd7a55a7416e91f8b511d21dfce671240630320f/software/o_c_REV/platformio.ini#L41C1-L41C42

By removing that line, Im able to build without any problems

djphazer, 2023-08-10:
Oof! Well that's not critical - it's a helper script that sets up the version string and filename. I've customized it to work for me on Windows... I'd still like to make sure it works in all cases.

What platform are you on? Can you provide the relevant error text?

jtomasrl, 2023-08-10:
sure, Im running Windows 11, PlatformIO Core, version 6.1.7 and got this error

```
Processing main (platform: teensy@4.17.0; framework: arduino; board: teensy31)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
'tail' is not recognized as an internal or external command,
operable program or batch file.
CalledProcessError: Command 'tail -n1 'OC_version.h'|tr -d '"'' returned non-zero exit status 255.:
  File "C:\Users\...\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "C:\Users\...\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\...\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "E:\modular\o_c\O_C-BenisphereSuite-phazerville\O_C-BenisphereSuite-phazerville\software\o_c_REV\resources\progname.py", line 5:
    version = subprocess.check_output("tail -n1 'OC_version.h'|tr -d '\"'", shell=True).decode().strip()
  File "C:\Users\...\.platformio\python3\lib\subprocess.py", line 424:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\...\.platformio\python3\lib\subprocess.py", line 528:
    raise CalledProcessError(retcode, process.args,
============================================== [FAILED] Took 0.44 seconds ==============================================

Environment    Status    Duration
-------------  --------  ------------
main          FAILED    00:00:00.438
======================================== 1 failed, 0 succeeded in 00:00:00.438 ========================================
```

djphazer, 2023-08-11:
Finding myself stumped with Python at the moment, I'd recommend installing Git Bash or similar unix-tools package for Windows to cover this missing dependency. Besides, they're incredibly useful developer tools ;)

commit 55d8bec8 (2024-10-21): Directly generate progname without calling bash scripts (#105)

Resolves #19
closed 2024-10-21 by djphazer