djphazer/O_C-Phazerville #20: Presets Not Remaining after Power Cycle

wlellington, 2023-08-10:
Presets saved through preset memory menu will only persist while module is still powered, but will be cleared/lost when module loses power/system is turned off.

This was tested on several current builds (default, stock1, stock2) and on flipped and non flipped versions.

Behavior exists when powered by modular power supply, or just via USB.

djphazer, 2023-08-10:
The Presets are only stored in RAM until you initiate an **EEPROM Save** via the right encoder long-press (twice)

wlellington, 2023-08-11:
Good to know! must have missed that in the docs!

djphazer, 2023-08-11:
No worries! It's not very well documented yet, so thank you for reporting your frustrations - I'm sure others will face the same confusion. And SynthDad's new video didn't quite explain it well either.

I'll take some time to make a wiki page and/or YouTube short as a resolution.

djphazer, 2023-08-15:
Hopefully this is explained more clearly on the Wiki now - https://github.com/djphazer/O_C-BenisphereSuite/wiki/Hemisphere-Config

I'll still try to make more video documentation when I can.

closed 2023-08-15 by djphazer