djphazer/O_C-Phazerville #21: Calculate App - Normalize clock signals for Random

snydoodlsh, 2023-08-15:
I've been trying to use the "Random" setting on the calculate app. It would be nice when using 2 random generators, the 2nd one just automatically adopts the clock of the 1st one. Maybe you could update the programming to automatically adopt the clock / trigger of the 1st one (normalized to the 1st trigger) and if there is a need to use a separate clock or have a different trigger, it breaks the normalization at trigger 2. 

I know I can use a stack cable to do the same things... but I'm lazy. Otherwise, I have to essentially split my signal into 2x in order to get 2 different random outputs. It seems unnecessary. I wonder if there are other apps where this normalization of clock in trigger 1 would be beneficial?

djphazer, 2023-08-15:
I agree wholeheartedly, and this won't be difficult.

closed 2023-08-15 by djphazer

snydoodlsh, 2023-11-08:
Thank you! I'm going to update my O_C this week with this update!