djphazer/O_C-Phazerville #23: ebb&lfo feature proposal

pyerbass, 2023-08-29:
Hey thank you for this incredible app!
would it be possible/interesting to add the following feature:
an extra line of option that let the user choose between the 3 tides modes: looping (as is now) ; trigger function; gate hold
the two new modes would use the reset input as a trig/gate input. 

Thanks

djphazer, 2023-08-29:
I've wanted this myself, but haven't successfully hacked the code quite yet. We might consider making a separate applet for the non-looping functionality...

djphazer, 2024-06-09:
Oneshot mode (AD envelope) is available in v1.7.1, using reset (TR2) as the trigger.

Eventually, I'll try to tackle the AR envelope mode with proper sustain (gate-hold as you call it)... leaving this issue open as a reminder.