djphazer/O_C-Phazerville #24: vector LFO bug

X-modular, 2023-09-24:
if the LFO is set to something slower than 0.10hz such as 0.02hz, modulating the lfo speed with CV1 will change the speed, but the LFO wont return to the slow 0.02hz setting after the modulation has returned to zero.  I think the new unwanted baseline for speed goes back to 0.10hz after CV1 has been twiddled.  The only way to get the speed back down to 0.02hz after it has been modulated is to adjust the speed internally.

djphazer, 2023-09-24:
I have noticed some weirdness around the slower frequencies... there wasn't really a clear case in the code that returns the value to normal after being modulated by CV1. I'm trying out a fix right now.

djphazer, 2023-09-24:
Pretty sure I've fixed the frequency modulation settling back to 0... watch for the next release.

I'm still getting weird behavior for many of the waveforms if it's set to 0.03hz or lower. It seems like certain waveforms (with vertical line segments, perhaps) get stuck in a high frequency alternating positive/negative loop. I suspect some kind of rounding error somewhere in `HSVectorOscillator.h`. Shapes like the sine wave work ok, though. What kind of waveshapes are you using?

Just be warned: this algorithm might only be reliable at 0.04Hz and up!

djphazer, 2023-09-26:
After fighting with this a bit more, I've concluded that the Vector Oscillator algorithm wasn't designed for speeds slower than 0.10Hz. For now, I've set the minimum for the LFO applet to 0.08Hz, which is the lowest stable frequency that worked for me with all waveforms.

We might achieve slower rates in the future with a substantial overhaul of `HSVectorOscillator.h` - until then, I'd recommend **Ebb&LFO** for extremely long cycle times. I know it's not the same as having custom waveshapes... leaving this issue open as a goal.

X-modular, 2023-09-26:
Okay.  Thanks for the effort.  It sounds like a complicated app to modify.  Ebb and LFO is nice for the long cycles.  Vector LFO is better for waveshape variety.  Thanks again.

athousanddetails, 2023-10-22:
@djphazer you know what would blast ? 

Make VectorEG just like Zadar. Uni-Polar, Cycling or Not, Envelope Generator... and preload tons of waveforms just like zadar :) And of course, make it as slow as possible, and as fast as possible. 

I like the Zadar implementation for Speed, it's done in ms / s / minutes which gives you a way better sense of what you are going! 

Cheers mate, awesome work!

djphazer, 2025-12-05:
The VectorOsc engine was overhauled by @qiemem a while back, and **VectorLFO** can go extremely slow now, just like **Ebb&LFO** - many seconds or minutes. Pretty sure this Issue is resolved, but let me know if something still needs attention.