djphazer/O_C-Phazerville #25: Cannot get to clock setup screen in 1.6.4

xmacex, 2023-09-25:
Hiya, with the new #2cc3ca0 (erm is that how do tag a release on GitHub?) I cannot anymore access the clock setup with the <kbd>▴</kbd><kbd>▾</kbd> gesture. Instead, it gets me to the new screen saver screen. It worked earlier today before I updated.

Looking at commit messages, #389790d has a hint maybe... can I help debugging/testing?

O_c is so cool! Thanks for w/forking on it 😎

djphazer, 2023-09-25:
Lol oh dammit, maybe we've taken a step backward. You've correctly pinpointed the commit in question. I'll have to revisit it...

I think the edge case was happening when you hit both buttons at exactly the same time - the first one is processed and the second (button down) is immediately ignored. When you release, the first is ignored but the second (button release) still fires, thus cancelling the screen.

With the old behavior restored, the trick is just pressing both buttons within a 60ms window.

xmacex, 2023-09-26:
The gesture I embodied with 1.6.3. (which is when I got onboard this Eurorack madness, with O_c being my second module and an obvious choice, not least because of this firmware) was to

1. press <kbd>▴</kbd>
2. press <kbd>▾</kbd>
3. (clock setup appears)
4. release <kbd>▴</kbd>
5. release <kbd>▾</kbd>

...a kind of a galloping 🏇 gesture. I admit I do it kind of quick, like one sweep left-to-right and it worked very well! It seems with 1.6.4. I fail to get into the clock setup because screensaver activates if I keep the <kbd>▴</kbd> pressed any longer. Lol I feel so clumsy. I'll work on my finger technique, and report back how it goes.

djphazer, 2023-09-26:
Just published v1.6.5 with the fix! Feel free to reopen if you still have trouble.

closed 2023-09-26 by djphazer

xmacex, 2023-09-26:
Just galloping :horse: here to inform that with 1.6.5. it works as it did in 1.6.3. Back at 145 bpm, the true and correct tempo haha :)

Cheers!