djphazer/O_C-Phazerville #27: Gate output from DualTM is not triggering ADSR module correctly

5-lagu, 2023-09-29:
Hey! I just tried using DualTM to trigger a Nano Modules Serra. I don't know if the gate input on this module is particularly finicky, but the gate output mode is not triggering the envelope correctly. The trigger mode seems to work fine. The gate mode in the Enigma Jr app and the other sequencers seems to trigger it as normal. I don't have a scope, so I'm not sure if I'm able to diagnose this futher much... I'll try another envelope module tomorrow and see if it makes any difference.

djphazer, 2023-09-29:
You have a Scope applet! ;)
The screensaver will also show you I/O status.

Some things to consider:
1. All outputs on DualTM are governed by the **Slew** parameter, which could be smoothing out the gate signals too much.
2. The Gate(1,2) output mode uses a combo of both registers for a 3-stage output value (0V, 2.5V, or 5V). The two registers are initialized as complements of each other, so this will yield a constant 2.5V until they are mutated.

5-lagu, 2023-09-30:
Cheers! I just found that the slew was set at 4 (maybe it's the default?), and reducing it to 1 was needed for the gate to trigger the Serra module properly.

djphazer, 2023-10-09:
I shall change the default Slew value to 1, or rather 0 to be clear ;)

5-lagu, 2023-10-11:
Good call. 0 for "no slew" makes the most sense in any case. :)

djphazer, 2023-10-24:
Slew defaults to 0 in v1.6.6

closed 2023-10-24 by djphazer