djphazer/O_C-Phazerville #30: TB-3PO Transposition Range Limited

cmovvv, 2023-10-13:
Hi - so I've got the most recent release of Phazerville (excellent work btw!) While operating the TB-3PO applet though, the transposition range is limited from C to F# across all available octaves. Not certain if this issue is related to your branch or Logarhythm's original code, but I couldn't find mention of it anywhere else. Is this something you have noticed or are able to replicate? Thank you for your time.

djphazer, 2023-10-13:
This has been bugging me, too... limited root note selection. It's still from the original applet behavior.

cmovvv, 2023-10-13:
> This has been bugging me, too... limited root note selection. It's still from the original applet behavior.

I'll contact Logarhythm then and see if they know how to fix this. Thanks!

djphazer, 2023-10-19:
I'm reworking this a bit... if I understand it correctly, up until now, the **Root** note was applied _before_ quantization, meaning every scale was effectively still in the key of C!

After my tweaks, **Root** note behaves as a semitone offset after quantization, which I think is more correct. And all 12 note values are available for all scales. **Transpose** will also act in scale degrees, while it previously behaved as a semitone offset (aka root note transpose).

djphazer, 2023-10-24:
Fixed in v1.6.6

closed 2023-10-24 by djphazer