djphazer/O_C-Phazerville #31: Could Scenes be 3 channels with 4th channel sending combined gate?

jococo, 2023-10-21:
Scenes, like Traffic, can be used to create drums patterns, for example feeding several Plaits voltages to make individual percussion sounds. In this situation the various gates need to be summed to create gate that triggers the Plaits envelope. This takes a fair bit of patching to achieve outside O_c but could be easily achieved by dedicating the last channel as the combined gate output. Traffic has this useful feature too. What do you think? Could Scenes be customised to achieve this?

djphazer, 2023-10-21:
Ah yes, I did consider this! When I get back to it, I think I'll probably make Left Encoder Long-press toggle output D to **Trig Sum** mode

jococo, 2023-10-21:
Awesome! Making it optional is an even better idea.

djphazer, 2023-10-24:
Implemented in v1.6.6

closed 2023-10-24 by djphazer

jococo, 2023-10-24:
Wow, that was quick work! Will check it out tonight after work.

jococo, 2023-10-25:
Works perfectly, thanks @djphazer !