djphazer/O_C-Phazerville #32: MidiIn - Polyphonic Gate may need further tweaking?

jpbed1, 2023-10-28:
First off...kudos for making the gate output respond properly to polyphonic MIDI input vs. the "stock" Hemi/Benispheres!
It is already a lot more functional as a MIDI to CV converter for this very reason. 
Perhaps we can fix CaptainMIDI too :D

One thing I noticed though was that the Note output is still a little quirky:

Current behavior:
- When pressing more than one key, the Note CV output correctly assigns the pitch as the last key triggering a Note On, 
- However, when releasing all but one of the keys, the pitch will only use the last detected Note On, whether or not that key is still being held down.

Expected behavior:
When there is only one key being held down, and other keys are released (Note Off), the Note CV should be assigned to whatever note is being held, which is not necessarily what note was pressed last.

Let me know your thoughts....JP

djphazer, 2023-10-29:
Indeed, it's still not truly polyphonic. It only keeps track of the most recent note and a count of NoteOn's. Eventually it will get more attention, when I tackle a use case that requires real polyphony, such as quantization with MIDI-controlled scale mask. ;)

beau-seidon, 2024-12-21:
over the next few days, i'll see about adding a note buffer which tracks the order in which they were played, and if they are released in any order, shifts any more recent sill-held notes down to prevent empty slots in the middle of the buffer. 

when the most recent note is released, the next-most-recent-still-held becomes the "active note," which facilitates a nice legato style performance. (picture a guy wearing leopard-print pants that are much too small, playing screaming keytar licks on a stage with big pyrotechnics.) 

i previously wrote some code that does this (the note buffering, not the pyrotechnics) and have used variations of it in a few applications and it works well. it should do what you want. 
i'll see if i can shoe-horn it in here without breaking too much. 

examples:
https://github.com/beau-seidon/TinyMIDI2CV/blob/main/software/src/cvnote.cpp
https://github.com/beau-seidon/NESizer2/blob/master/src/note_stack/note_stack.c

beau-seidon, 2024-12-22:
IT'S VERKING!!

https://github.com/beau-seidon/O_C-Phazerville/tree/midi-modz

next i'll see about configurable polyphonic mapping to the 4 outputs then put in a PR. please give it a test and see if i borked anything

djphazer, 2024-12-22:
Sweeeet, I'll check this out today!

beau-seidon, 2024-12-22:
increase POLYPHONY_MAX (HSIOFrame.h line 27) to 8 before you build it. it's better that way.

also, i think it would be better if the live printout of the midi note-on log didn't print duplicates for ADC channels that share a midi channel. i'll look into breaking that later.

beau-seidon, 2024-12-23:
the changes i made earlier remember velocity in the same way as the notes but i just found that it doesn't send vel to 0 when the last note is released. will fix

djphazer, 2024-12-23:
Very glad to have you giving this the attention it needs! A quick test looks promising.

I had previously debated whether using something like std::vector was a good solution for a note stack... but never got very far. Polyphony of 8 or more is desirable tho.

beau-seidon, 2024-12-23:
changing polyphony is easy, i initially chose 4 because 4 outputs, but during my first test i immediately realized that's ridiculous. it is much better with 8 and i don't see any reason to limit it other than the buffer eats (POLYPHONY_MAX * 16 * 2) bytes. the buffer is POLY_MAX notes per midi channel.

most people don't have more than 10 fingers so 8 is a decent compromise. 16 would be better for feeding it big arrangements from a daw or something, but again, the hardware only has 4 outputs (not sure about T4.1 specs). 

std::vector is dynamically sized right? i have always just used built in arrays for all my previous stuff, but i know you are trying to conserve memory for the new firmware rev..

djphazer, 2024-12-23:
Actually, I'm mostly trying to optimize and conserve resources for the original Teensy 3.2 processor. T4.x can handle things quite easily. I enjoy the challenge of optimizing things to run on the lesser hardware ;)

outside of static arrays - which are probably ideal for small systems like this - vector is the standard dynamic list thing to use, but might have tradeoffs...

beau-seidon, 2024-12-27:
the latest commit on my [midi-modz](https://github.com/beau-seidon/O_C-Phazerville/tree/midi-modz) branch updates the midi note/vel buffer to use std::vector.
should behave the same as before from an end-user perspective, but with better memory management internally.
the output is still monophonic for the moment, but the buffer is no longer limited to a static size, and when it goes empty (last note released) it frees up the allocated memory so other apps can use it.

it took me a while to get it working the way i wanted tonight. i realized what i was doing was way overkill and was able to drastically simplify everything i wrote. ie. i pretty much deleted most of it. the major hangup was erasing elements from the middle of a vector, because im feeding it structs which combine the midi note number and velocity. [this ](https://en.wikipedia.org/wiki/Erase%E2%80%93remove_idiom) is what i needed to know in order to fix it

it's working great on my end, but let me know if you find any boo-boos, or if i did stuff you don't like to your code.

beau-seidon, 2024-12-27:
> also, i think it would be better if the live printout of the midi note-on log didn't print duplicates for ADC channels that share a midi channel. i'll look into breaking that later.

updated this ^ in latest commit. 

i also just noticed the MIDI activity icon which blinks at the top of the screen when O_C receives midi messages overlaps with the applet title on the right HS. 

i think it might be neat to make it blink next to the channel selectors, if the incoming data is for that particular channel. 
but for now i'm just trying to figure out how to make it mirror the left HS. how does an applet know if it's on the left or right side? is that a thing?

djphazer, 2024-12-27:
@beau-seidon You're on fire, my friend! Happy to see you digging in deeper. We definitely need to tweak that activity icon...

Every applet has a `hemisphere` member variable, assigned 0 for left, 1 for right side, with enum values defined for convenience - `LEFT_HEMISPHERE` and `RIGHT_HEMISPHERE`. Newer expanded hardware runs _four_ applets simultaneously, but still only two visible at a time... even numbers on the left, odd numbers on the right.

An efficient check for which side you're on would be:
```
if (hemisphere & 1)
  rightSideStuff();
else
  leftSideStuff();
```

beau-seidon, 2024-12-28:
sweet thank you. icon overlap fixed by latest commit. 
didn't implement the per channel monitor, but its on my list. just moved the icon over on the right HS for now.

beau-seidon, 2024-12-28:
got polyphony working! 

i added 4 new modes to the MIDIFunctions enum: POLY1, POLY2, POLY3, POLY4, which writes the corresponding last 4 notes that were played to the outputs with these functions selected. POLY1 is the same as NOTE_OUT, wasn't sure if i should include that or not because it's redundant.

the issue i'm having now is config retention after restart. even if i save the state using a right-encoder long-press, it comes back with different config after restart. gotta go find out how saving works.

djphazer, 2024-12-28:
This is exciting! Can't wait to try it out with a 4-voice drone...

For saving, we need to expand beyond 3 bits for the selected function. How many other modes might we add? will there be more than 32 modes? if not, 5 bits is probably good. Something like this:

```
diff --git a/software/src/applets/hMIDIIn.h b/software/src/applets/hMIDIIn.h
index 7cddb972..796a6f5e 100644
--- a/software/src/applets/hMIDIIn.h
+++ b/software/src/applets/hMIDIIn.h
@@ -116,18 +116,20 @@ public:
         uint64_t data = 0;
         Pack(data, PackLocation {0,4}, frame.MIDIState.channel[io_offset + 0]);
         Pack(data, PackLocation {4,4}, frame.MIDIState.channel[io_offset + 1]);
-        Pack(data, PackLocation {8,3}, frame.MIDIState.function[io_offset + 0]);
-        Pack(data, PackLocation {11,3}, frame.MIDIState.function[io_offset + 1]);
+        // 6 bits empty here
         Pack(data, PackLocation {14,7}, frame.MIDIState.function_cc[io_offset + 0] + 1);
         Pack(data, PackLocation {21,7}, frame.MIDIState.function_cc[io_offset + 1] + 1);
+
+        Pack(data, PackLocation {28,5}, frame.MIDIState.function[io_offset + 0]);
+        Pack(data, PackLocation {33,5}, frame.MIDIState.function[io_offset + 1]);
         return data;
     }

     void OnDataReceive(uint64_t data) {
         frame.MIDIState.channel[io_offset + 0] = Unpack(data, PackLocation {0,4});
         frame.MIDIState.channel[io_offset + 1] = Unpack(data, PackLocation {4,4});
-        frame.MIDIState.function[io_offset + 0] = Unpack(data, PackLocation {8,3});
-        frame.MIDIState.function[io_offset + 1] = Unpack(data, PackLocation {11,3});
+        frame.MIDIState.function[io_offset + 0] = Unpack(data, PackLocation {28,5});
+        frame.MIDIState.function[io_offset + 1] = Unpack(data, PackLocation {33,5});
         frame.MIDIState.function_cc[io_offset + 0] = Unpack(data, PackLocation {14,7}) - 1;
         frame.MIDIState.function_cc[io_offset + 1] = Unpack(data, PackLocation {21,7}) - 1;
     }
```

With 64 bits to work with, I moved the bit position rather than resizing in place, so that loading old data will be less weird.

beau-seidon, 2024-12-28:
i hope your intention was for me to just copy/paste that. because that's what i did, and saving works now. thank you!

beau-seidon, 2024-12-28:
> How many other modes might we add? will there be more than 32 modes?

i can think of a several more that would be useful, but no probably not more than 32.

in my dual-output TinyMIDI2CV application i have a bunch of different modes for my 2nd CV output. i might port a few of them into o_c because they are very fun performance utilities. 

for example, "keyboard invert" swaps the output from the high and low keys. 
ie: `v_oct = cvMap(127 - note#)` which makes for some very interesting harmonies even with monophonic MIDI input, when played together with the normal v/oct CV into a 2nd VCO. 
also there are several para/duophonic modes, like low, high, last, pedal, outer, etc., and a couple different retrigger modes.

i'll probably put in a PR soon to close this particular issue, if you don't find any nasty bugs while you're playing with it or have any other feature requests or recommend changes.
i'll try to add the other stuff i mentioned later and i'll look into some of the other issues on here too. i want to get to know the o_c better, this platform is amazing!

closed 2025-04-04 by djphazer