djphazer/O_C-Phazerville #33: Thank you for taking O_c forward

xmacex, 2023-10-29:
The original Ornament and Crime was already a very interesting, and inspiring /inspired, and importantly affordable open source module. The Hemisphere took it to a new direction and the modularity idea there was so good.

Your Phazerville firmware carries forward (via Benisphere, respect there too) and adds onto the legacy of this legendary module.

[image]

I am convinced I speak for many O_c users, no matter which firmware they use, in the following: thank you for what you have done to maintain and develop Ornament and Crime.

Have a nice day 🕊️

[To be closed after you've read this]

djphazer, 2023-11-02:
Thank you so much for your kind words! This has been a very rewarding hyperfixation - moreso than actually producing music. People tend to be far more interested in music tools than random beats some guy made.

Software Development is my art 🎨

Stay tuned - the best is yet to come!

closed 2023-11-04 by djphazer

xmacex, 2023-11-04:
Software development is a legit artform! Cheers :)