djphazer/O_C-Phazerville #35: O_C Settings Bug?

snydoodlsh, 2023-11-22:
I was looking over the calibration settings on 1.6.6 and tried changing the knobs on my Plum Audio O_C so when turned clockwise each app advanced to the next app (alphabetical progression) instead of going "backwards". Trying the "reverse for both" only reversed the knob direction on the left knob. I finally got the setting I wanted by selection "reverse left". For some reason, that reversed both knobs. Not a big deal, but I thought you should know.

djphazer, 2023-11-23:
The wording of that setting is somewhat arbitrary and hardware independent. The original 14HP panel used the "normal" configuration, while the 8HP panel has one of the encoders reversed. So, the setting is not how the encoders function relative to each other, it's what is needed to make them agree with each other on your particular hardware variant.

I might overhaul the whole calibration routine at some point, and when I do, encoder calibration will be something simple like "Turn the each encoder clockwise"

ph1-xyz, 2024-08-11:
If I may add, it would be nice to add the encoder configuration to build flags, as it is FLIP.