djphazer/O_C-Phazerville #37: Feature request: Trig/gate on quantiser not change

kaffeetrinken, 2023-12-16:
Hi Nick   :) 

I wanted to check if it would be possible with out major efforts to add a gate high / trig on note change to any of the existing quantisers in hemispheres? It's the one of the things I'm really missing from Meta-Q. (Use case: Firing an accent event on note change to simulate a blow of sorts you would have in wind instruments …).

Many thanks. And thanks for all the great work so far — it's a game changer!

djphazer, 2023-12-16:
This is an excellent idea! And it won't be very difficult. I've previously used Meta-Q for similar functionality, and would immediately enjoy having this in an applet.

djphazer, 2023-12-28:
I found myself craving this last night... just pushed an update to ScaleDuet - https://github.com/djphazer/O_C-BenisphereSuite/commit/1e7e45fc6f15f95882552396e9bc824027b51e13

You can grab a fresh build from the [workflow run](https://github.com/djphazer/O_C-BenisphereSuite/actions/workflows/firmware.yml), or tell the bot to regenerate your own Custom Build ;)

djphazer, 2024-03-22:
This is now available in v1.7 with ScaleDuet. I hope that works for you!
I'm fond of this feature, so it may sneak into some other quantizer applets in the future...

The new 8-channel T4.1 hardware will also have some extra functionality in Hemisphere - auto-triggers on the extra outputs when the main pitch outputs change. ;)

closed 2024-03-22 by djphazer