djphazer/O_C-Phazerville #40: Update version

lucianthorr, 2023-12-20:
Hello! Should the version be updated in[ OC_version.h](https://github.com/djphazer/O_C-BenisphereSuite/blob/1829dd4ea9c32b7cfeba8343a96a46faf60a0999/software/o_c_REV/OC_version.h#L2)?

I just loaded the firmware and the Setup/About section says it's still on v1.6.6.  It looked like a successful update so I'm guessing it's just coming from here.

And thanks for everything you're doing here.  It's a real joy to follow and try to keep up.

djphazer, 2023-12-20:
Double-check the filename you opened with the loader... and you've somehow linked to an older version of that file - https://github.com/djphazer/O_C-BenisphereSuite/blob/phazerville/software/o_c_REV/OC_version.h

lucianthorr, 2023-12-20:
Ahh, I always forget Github search will give results from all branches.  I just assumed that was off the default.
You're right, it all installed correctly on the second try so it must've just been a random failure or late night user error.
Thanks

closed 2023-12-20 by djphazer