djphazer/O_C-Phazerville #43: voltage drop when screensaver appears

sebpmtl, 2023-12-30:
Happened when I was using SequenceX, I noticed a slight voltage drop when the screensaver appeared. I could hear it on the sequence playing.  today, I'll re-test with my oscilloscope to see what's going on.

djphazer, 2023-12-30:
VOR? Auto-save enabled?
Auto-save will cause a slight hiccup...

If it's really a voltage drop from extra processing, try the `[blank]` screensaver.

sebpmtl, 2024-01-02:
also, saving doesn't work anymore. I save as I always saved and it starts on DualTM and EuclidX on startup.

sebpmtl, 2024-01-02:
I did't see the extra settings. Sorry. I think it was my quantizer.

djphazer, 2024-01-05:
Hopefully this was just a false alarm? It seems like a slight possibility on VOR hardware, but I wasn't able to reproduce. Feel free to reopen if you're able to nail it down.