djphazer/O_C-Phazerville #44: Golang errors building

lucianthorr, 2024-01-06:
I thought I'd give writing a little Hemisphere Applet a shot because this PlatformIO environment seems so approachable but I'm running into some scary errors right from the start.

I'm guessing these aren't O_C-Phazerville specific but if you have any ideas, that would be great.

I'm on a M1 Mac, using the VSCode PlatformIO plugin, I run `oc_stock1`> Build and get errors compiling Go dependencies:
```
Assembler messages:
Error: unrecognized option -mno-unaligned-access
*** [.pio/build/oc_stock1/src/pkg/mod/golang.org/x/sys@v0.14.0/cpu/asm_aix_ppc64.s.o] Error 1
```

Do you know if anyone is successfully building on Apple Silicon machines?

lucianthorr, 2024-01-06:
Weird.  I removed the `pkg/` folder containing all the Go dependencies and now everything builds perfectly fine.
No idea how that got in there.  I'll keep a closer eye out as I try to learn more.

closed 2024-01-06 by lucianthorr

djphazer, 2024-01-07:
Very strange! AFAIK, nothing in this project uses Go. Maybe you have some other VSCode plugin that pulled it in? I don't fully understand PlatformIO, but I do know that deleting the cached folders can help, as you discovered.

Good luck with your applet development!