djphazer/O_C-Phazerville #48: Save not functioning in v1.6.999

ericneilj, 2024-01-14:
Using version o_C-phazerville-v1.6.999_phz-d0367980.hex
Standard Ornament and Crime module with teensy 3.2

Press and holding right encoder brings up app selection screen, press and holding right encoder until "Saving..." and progress bar animation executes. Power module off then on, default on-screen apps are DualTM and EuclidX regardless of apps selected prior to saving

djphazer, 2024-01-14:
You have to store to a Preset in Hemisphere. I did have a mechanism to automatically save to Preset 'A' but things changed a bit under the hood...

djphazer, 2024-01-21:
I think I'll be revising the way this works in v1.7. An invisible Preset Zero could hold the active state, restoring former behavior.

Currently, you must use the [Hemisphere Config](https://github.com/djphazer/O_C-Phazerville/wiki/Hemisphere-Config) screen to either Save to a Preset slot manually, or enable Auto-Save and then activate screensaver / exit to main menu.

ericneilj, 2024-01-22:
thanks for the clarification @djphazer

djphazer, 2024-03-22:
This should be resolved in v1.7 - Preset A is always automatically stored when retreating to the App menu with a long-press.

closed 2024-03-22 by djphazer