djphazer/O_C-Phazerville #51: Calibr8 applet displays with calculate behind

yratof, 2024-02-06:
[image]

Not sure the bug here, but Calibra8 overlaps calculate on latest

djphazer, 2024-02-08:
Not a bug - Calibr8 just uses the same "skyline" visual to easily see the inputs & outputs. The applet itself was a proof-of-concept that I didn't intend on developing further... I removed it once, but at least one person out there wanted it back. Feel free to improve it with a PR.

beau-seidon, 2024-12-20:
[image]

do you like this better? if so, i will put in a PR. it's just a small change to Calibr8.h

beau-seidon, 2024-12-20:
i might change it again, and give each section (A, B | C, D) its own pair of bars beneath the numbers instead of all bars at the bottom.

beau-seidon, 2024-12-20:
[image]

yea, bars in each section is definitely better

djphazer, 2024-12-20:
I like it! Maybe even get rid of the divider line.

beau-seidon, 2024-12-20:
will do tomorrow evening. that will let me spread things out a few more vertical pixels, and prevent collision with the cursor.

beau-seidon, 2024-12-21:
[image]

middle line removed and other ui element spacing adjusted very slightly to clean everything up.

updated meter bar behaviour to handle negative input and output values:

if the corresponding value is positive the bar extends from the left side toward the right as the value increases. if it is negative, the bar now "wraps" like a pacman maze and goes from the right side toward the left as the value goes more negative.

the scale factor and offset parameters adjust the relative size and position of the bars.

PR #111 (2024-12-30): Calibr8 UI update
fixes #51 

the background Skyline visuals have been replaced with horizontal bars below the numbers in each section indicating cv input and output levels.

middle dividing line has been removed and other ui element spacing adjusted very slightly to clean everything up.

input and output meter bars can handle both positive and negative values:
if the value is positive, the corresponding bar extends from the left side of the segment toward the right as the value increases. 
if it is negative, the bar "wraps" like a pacman maze and goes from the right side toward the left as th [... cut]
closed 2024-12-30 by djphazer