djphazer/O_C-Phazerville #53: posessed O_C - fluctuation voltage on phazerville but calibrateable and stable on Benisphere

tastez, 2024-02-22:
hi - bizarre question but maybe someone ran into this - i've got an old diy O_C that functions totally fine but after installing Phazerville 1.6.999 or 1.7.0 the outputs are just fluctuating like crazy. I thought its a hardware issue but it ends up stabilizing on the Benisphere Calibration screen and you can calibrate it without issue and everything is great. Once you install phazerville is spazzes out again and calibration screen or process wont make it stabilize. I've installed it on another O_C and there is no fluctuating and I didnt need to calibrate it, but will try. I guess something is stuck somewhere but reboots and reuploads with the teensy hex uploader dont fix it. Weird!

tastez, 2024-02-22:
nah this is definitely something with the hardware. somethings failing after a bit or randomly

closed 2024-02-22 by tastez

djphazer, 2024-02-22:
Well now I'm curious... is it bad calibration data in EEPROM? You're not using a VOR build, right?

Maybe try doing a full reset on that module. Let me know if you figure it out.