djphazer/O_C-Phazerville #54: Feature request: Triggered changes to ProbDiv and ProbMeloD settings

gabriel-roth, 2024-02-25:
Often when people post feature requests they say "I think this is a small request that could be added without much trouble." That is not the case with this request. 

I love using ProbDiv and ProbMeloD together — they are among the best tools I've found for randomly generating sequences that feel musical. Last night I got into an epic jam and wanted to do two specific things that aren't possible with the applets in their current state: 

(1) every four bars, force ProbD to land on /1, to hold a note for a while
(2) make ProbMeloD periodically "change chords" by switching to a different set of note probabilities. 

Both of those — and lots of other useful things — could be achieved by allowing the applets to store multiple "probability sets" and switch among them via CV, akin to changing scales in a quantizer or changing Turing Machines in Enigma. At the moment, each only has a single probability set, which can only be changed manually. 

Obviously, implementing my request would require an interface for saving, editing, and switching probability sets, and for CV assignment. It may be that the way to achieve it would be to create a new app, outside Hemispheres, that would combine the ProbDiv and ProbMeloD functionality and add the ability to change probability sets. 

Like I said, not a small request, but I thought I'd post it here and see if it attracts any interest. 

Thanks again for your fantastic work.

gabriel-roth, 2024-02-25:
Oh, that's interesting — I hadn't thought of using Presets because I don't think of them as being switchable on the fly. (I don't use MIDI so I wasn't aware that they can be changed that way.) I would love to be able to control them with CV but I can see how that's a challenge since you don't know what the CV inputs are being used for …

russb78, 2025-05-08:
There are ways built in to ProbMeloD to change the melody in a stochastic way - they're off to the right (After the B key). That said, I have no idea how they work since I can't find any more information on them. 

The biggest issue with me messing around with them is that it doesn't honor your chosen scale / ignore notes that have 0 probability. 

If Prob Melod respected the scale / mode you've selected in the Config (or push an offending note a semitone up or down so it stays in key) then that would be amazing. 

If that's too hard, maybe replace all the extra undocumented stuff off to the right with a simple up / down scroll between up to 4 / 8 melodies. maybe a 5V to the reset (instead of 2.5v) could cycle through them.