djphazer/O_C-Phazerville #55: o_C-phazerville-v1.7_T40-6d32547c.hex freezes on powerup.

juanlittledevil, 2024-02-27:
I'm currently unable to get the 1.7 build with a teensy 4.0 to work. I've confirmed that my hardware supports the 3.2 without any issues but the 4.0 builds freeze on the startup splash screen. I've tried with a few different boards and two modules (albeit same pcb layout) and still get a frozen screen after a few seconds post boot, then a few seconds later the screen just goes black. There is also a faint clicking I can hear from the microcontroller. Not sure if anyone else is having similar issues.

djphazer, 2024-02-27:
Faint clicking? 🤔
Tell me a little more... does the splash screen animate at all before it freezes? Is it DIY hardware or from a vendor? Did you cut the USB power trace on the Teensy?

The Serial Monitor inside Arduino IDE might be able to provide clues and a CrashReport, if you have it.

juanlittledevil, 2024-02-28:
yeah, on a timed interval. I almost didn't catch it because it's really quiet.

The screen does animate but only sometimes and very briefly. The hardware is DIY and yes I did cut the USB power trace on the Teensy.

I don't use the Arduino IDE, but if you can point me to instructions on getting a CrashReport, I'm happy to produce one.

djphazer, 2024-02-28:
Ok, strange... looks like it almost worked, but then it freezes. Maybe it's a driver bug, or possibly garbage data in EEPROM.

It might not show much, but to check the Serial output, you'll need the latest [Arduino IDE](https://www.arduino.cc/en/software) for your system + the [Teensyduino extension](https://www.pjrc.com/teensy/td_download.html).
Once installed, you can connect to a Teensy via USB and then view the "Serial Monitor" from the Tools menu. Once connected, it should automatically reconnect when you power cycle the module and possibly spit out crash reports...

If it is still alive while the screen is frozen, you might be able to use Paul's Screen Capture utility to investigate.

juanlittledevil, 2024-02-28:
Sadly, after installing the Arduino IDE (2.3.2) and the latest teensyduino extension. Nothing gets dumped to the Serial Monitor. The tool does see the board and port as I see a message saying: Message (Enter to send message to 'Teensy 4.0' on 'usb:2130000') But nothing gets uploaded. I see on the Main.cpp that serial is configured with 9600 baud rate which is also what is set on the Serial Monitor. So likely the app is crashing before anything is printed, unless there is an additional flag that must be set to enable debug.

Your theory on this being a bug with the driver seems more plausible. This behavior seems very much like a memory leak, but not really knowing this code I would not even know where to begin to look for the problem.

djphazer, 2024-02-29:
Thank you for taking the time to debug this with me. Glad to see you found the PRINT_DEBUG flag as well! It is very helpful, though CrashReports will still come through without it.

I have a few more ideas to isolate the problem:
1. Disable some or all of the app flags for the T4 env target in platformio.ini, see if it runs on a minimal build.
(Add the `-DNO_HEMISPHERE` flag to exclude Hemisphere)
2. Disable the -DTEENSY_OPT_SMALLEST_CODE flag
3. Try out an older v1.6.x build, maybe I broke something in 1.7

Meanwhile, I'll be looking for memory leaks...

juanlittledevil, 2024-03-03:
Sorry for the delay. Here are the results from my troubleshooting efforts:

1. Commented out -DENABLE_APP_* from [emv:T4] in the platformio.ini file, compiled:

No change, same outcome.

2. Commented out -DENABLE_APP_* and added -DNO_HEMISPHERE, the tested:

No change, same outcome.

3. commented out -DTEENSY_OPT_SMALLEST_CODE from [env], tested:

No change, same outcome.

4. Downloaded the following versions and tested:

[v1.6.777 Beta](https://github.com/djphazer/O_C-Phazerville/releases/tag/PSv1.6.777) - Different screen (boots with calibrator) but still frozen and I can hear the clicking. 
Ok so from what I can gather, the issue isn't with Hemisphere or the apps. So more likely that this means is we are narrowing down on a driver issue. Hmm.. I wonder If this is more of a timing issue.

juanlittledevil, 2024-03-20:
Ok, I've been troubleshooting this over the last few days and I think I found my issue and it does seem to be hardware-related. looks like the LM1117 is shutting itself off due to heat even though I had added what I thought was enough copper to dissipate heat. The regulator does get somewhat warm to the touch. But when I put my finger on it it operates longer. So... I put some thermal tape on the regulator and it had been operating for a few minutes before it crashed. Anyway, I'm going to rework the schematic to allow for more heat dissipation and test again.

djphazer, 2024-03-20:
Awesome, thanks for the follow up! Glad to hear it's probably not my fault! 😅

I suspected it could be hardware related... but the Teensy 3.2 still works fine? Hardware circuit details are a little lost on me, but I'm curious to know why or indeed if it only fails with the T4.0

juanlittledevil, 2024-03-20:
> I think it is because the teensy 4.0 draws more current so the regulator
is heating up more. I put a heat sync on it and it’s been running for an
hour now. Was looking at th datasheet closer and looks like I stupidly put
the 300mA version, if I had used the 800mA then it would have had more
headroom and not work as hard. Oh well lesson learned. I will be making
another round of boards I guess. Tho at this point I maybe will just hold
off Paul finishes the next version? ¯\_(ツ)_/¯

The hardware isn’t too complicated. Easier than analog circuits in most
regards. I’m happy to share my experiences with you if you ever want to
chat. Just message me privately.

closed 2024-03-20 by juanlittledevil