djphazer/O_C-Phazerville #61: O_C Phaserville 1.7 on Plum Audio OCP - Attenuverters and voltage outputs are off

GabrielCrecan, 2024-03-13:
Hi,
I'm running 1.7 on a Plum Audio OCP and have just calibrated the module, but when using any applet, the values I get from the outputs are off by ±0.1v to ±0.5v. Is this expected due to hardware limitations ?
Another issue is related to the attenuators; If using the 'Scope' applet and setting an attentuator to full CW position, the output voltage is 4.92 (and -3.25 CCW to +4.87 CW when bi/uni button is depressed, so in Biplolar mode)

Any help in understanding this would be appreciated, 
Thank you

djphazer, 2024-03-13:
It kinda sounds like it's not properly calibrated... When calibrating the CV inputs with 1V and 3V, make sure the attenuators are open all the way. Make sure they are closed for the 0V point.

Double-check the Asym Vbias calibration step to ensure proper DAC offset for the ±5V mode (Hemisphere default)

All that being said, there is a known issue that can cause a slight discrepancy between what you measure in the calibration routine and in normal operation, but in my experience, it's only about ±0.01V

I plan to overhaul the calibration routine eventually to make it more intuitive and potentially more accurate.

GabrielCrecan, 2024-03-14:
Thank you for the reply. 
Just a quick clarification: 

- 'make sure attenuators are open all the way. Make sure they are closed for the 0V point.' 

You mean they need to be set to Unipolar ? 
=> 
all the way closed = 0v, 
all the way open = 5v 

I presume that when they are in bipolar mode, fully closed means -5v and fully open +5v

djphazer, 2024-03-15:
Ah yeah, Unipolar mode. I was just throwing out a couple of ideas... I usually leave the attenuators in Unipolar mode because that's the only way I can be sure they're set to 0 volts (full CCW)

Are you using a precision multimeter to calibrate the DAC?

GabrielCrecan, 2024-03-15:
Got it, thank you.
I'm using the Disting EX's 'Show input voltages' feature from Settings -> Calibration.
Unit is assembled by Plum Audio.

Interestingly, when calibrating OCP and reaching the 0 V bipolar and asym step, the 4 outputs have slightly different values A and C are around -0.5, while B and D are close to 0

djphazer, 2024-03-22:
Did you find a resolution to this?
I was wondering if things work properly in Unipolar (0-10V) range. If calibration is good for Unipolar, but outputs are inconsistent in Asym or Bipolar, I'd suspect a problem with the Vbias circuit.

GabrielCrecan, 2024-03-25:
No, definitely several things are wrong. Did the following tests (after re-calibrating and checking the that calibration was not lost):

1. Attenuverters: Unipolar & fully CCW
Range:
5v to +5v:    A=-0.05, B=-0.01, C=-0.03, **D=0.00**
3v to +7v:    A=-0.03, B=-0.01, C=-0.02, **D=0.00**
0v to +10v:  All outputs = 0v

2. Attenuverters: Unipolar & fully CW
Range:
5v to +5v:    A=4.94, B=4.98, C=4.97, **D=5.00**
3v to +7v:    A=5.02, B=5.05, C=5.01, D=5.07
0v to +10v:  A=5.05, B=5.05, C=5.02, D=5.07

Checked all above test cases using 'AttenOff' (with the default settings) in Hemisphere (Double checked using 'Scope' and got same output values)

Interestingly, got slightly different values while testing the same cases in with the Calibr8or app (also with the default settings):

1. Attenuverters: Unipolar & fully CCW
Range:
5v to +5v:    same
3v to +7v:    same
0v to +10v:  same

2. Attenuverters: Unipolar & fully CW
Range:
5v to +5v:    same
3v to +7v:    A=5.01, B=5.07, C=4.98, D=5.09
0v to +10v:  A=5.05, B=5.08, **C=5.00**, D=5.08

I'm not even going to dive into what happens with the attenuverters set to Bipolar mode, but will mention that the 'Scope' applet displays -3.30v CCW to +5.02v CW on the left hemisphere and -3.32v CCW to +4.97v CW on the right hemisphere.

I understand some could see this as nitpicking, but honestly it's driving me nuts.
Perhaps there's no point in seeking more precision that the above values ?

djphazer, 2024-04-01:
Hey, sorry for such delayed response... the variance does seem to be more than I'd expect, but if we're just nitpicking the max voltage that's normalled to the attenuverters, I wouldn't be surprised if it's not exactly 5V. Looks like some of mine max out at 5.02V... unless my ADC calibration is a little off.

I would verify ADC and DAC calibration independently, using _Scope_ with external voltage source input, and _Voltage_ for outputs to measure.

closed 2025-12-05 by djphazer