djphazer/O_C-Phazerville #68: Bug - Clock Skipper CV

jamesdmorgan, 2024-04-13:
Noticed a problem with the cv input for Clock Skipper.

The documentation says it modulates the probability.

I found that it didn't do anything. I validate the CV input on other hemisphere apps and it works fine.

Unless i'm doing something completely wrong I think it doesn't work.

Cheers

James

djphazer, 2024-04-13:
The input modulation probably works, but it looks like the UI just doesn't reflect the modulated value. I'll definitely fix that! Thanks for reporting this!

jamesdmorgan, 2024-04-13:
Nice one. I ran it through Mordax data and didn’t see a change in gates but
I may have given up

It definitely changes when I messed with the encoder.

Btw your additions have really enhanced O_c massively

Nice work

djphazer, 2024-06-09:
Fixed in v1.7.1

closed 2024-06-09 by djphazer