djphazer/O_C-Phazerville #69: Applet Suggestion: Nibbler-esque binary adder

SavageMessiah, 2024-04-19:
A fun thing you can do with the Schlappi nibbler is to generate gate patterns by using the adder. It's really simple and seems like it could be a fun applet for hemispheres. The user would choose a value to add/subtract with each clock, the number of bits in the adder, and choose two bits to route to the outputs. If we're feeling saucy you could also choose to output a stepped voltage, too. Given what a lot of the other applets are like, this seems like it would be pretty easy to implement.

So something like this:

TR1: clock TR2: reset
CV1: offset to add/subtract value CV2: offset to bit length ?
OUT1: configurable OUT2: configurable

The output choices would be a bit number or stepped voltage.

A fancier version of the applet would let you choose from pairs of bits and a boolean function on them for each output but I'm not sure if that's worth the additional UI complexity. It would still be less complex than DualTM so maybe that's fine?

zerbian, 2024-04-27:
Looks like an interesting idea, I might look into it and create a simple demo.

zerbian, 2024-04-29:
I did get some time to spend and worked on this little applet ("Cumulus"):
[image]

You can set the operation mode (ADD, SUB, MUL, XOR with ROTL) and assign the different bits of the accumulator register (Z) to the outputs. The constant value (k) can either set by the encoder or the CV1 (this overwrites the encoder value).

It tested it and it gave some interesting rhythms, especially when modulating the constant. I might look into more interesting operations and maybe implement some stepped voltage output as well.

https://github.com/zerbian/O_C-Phazerville/tree/app-accumulator

ph1-xyz, 2024-04-29:
> I did get some time to spend and worked on this little applet ("Cumulus"): 

What does your clock icon do? If you may like, I dabbled a few minutes with the pixel editor :-D

{ 0x02, 0x11, 0x21, 0x22, 0x44, 0x12, 0x42, 0x84 }
{ 0x49, 0x26, 0x08, 0x54, 0x54, 0x24, 0x8a, 0x01 }

zerbian, 2024-04-30:
The clock icon is just there to represent the current operation (as label), but I probably can get rid of it.
Are your icons supposed to be clouds? I cannot not really recognize their shape.

djphazer, 2024-04-30:
Haha they do seem like vague cloud-esque textures... do they tile? Could probably squeeze them into the header 😜

Happy to see this applet though, just in time to slip it into the next beta build for testing!

zerbian, 2024-04-30:
Sounds like a plan! I am going to fix some stuff I noticed, implement some parameter saving and then prepare a pull request.

ph1-xyz, 2024-04-30:
Something like...

{ 0x20, 0xa4, 0x12, 0x12, 0x89, 0x61, 0x12, 0x10 }
{ 0x40, 0x10, 0x88, 0x84, 0x44, 0x48, 0x24, 0x10 }
{ 0x10, 0x28, 0x48, 0x90, 0xa0, 0x20, 0x58, 0x04 }

[image] [image] [image] [image] [image] [image] [image] [image] [image] 

or, another idea could maybe to cycle through them as an animation?

zerbian, 2024-04-30:
Originally I choose the clock icon because I dint find any icon that corresponds to "Operation" or "Mode", so I choose the clock as the instruction only happens on the clock pulse. If you have a better idea, feel free to design something @ph1-xyz

ph1-xyz, 2024-04-30:
I may be understanding now, it regarded the operation at the right and not something regarding clocking...

An idea that comes to my mind is that of the [beaker](https://en.wikipedia.org/wiki/Beaker_(laboratory_equipment)), which, boiling will produce more or less (disordered) bubbles.

So, the icon could represent the bit-"entropy" produced from the app :-)

{ 0x70, 0x88, 0x07, 0x20, 0x00, 0x17, 0x88, 0x70 }
[image]

{ 0x70, 0x88, 0x27, 0x00, 0x90, 0x07, 0x88, 0x70 }  
[image]

{ 0x70, 0x88, 0x27, 0x40, 0x90, 0x27, 0x88, 0x70 }
[image]

ph1-xyz, 2024-04-30:
May we even try different shape per operation? :-D

{ 0x02, 0xfe, 0x00, 0x48, 0x80, 0x10, 0x00, 0xfe }
[image]

{ 0xc0, 0x20, 0x9c, 0x00, 0x40, 0x9c, 0x20, 0xc0 }
[image]

zerbian, 2024-05-01:
see https://github.com/djphazer/O_C-Phazerville/pull/72

djphazer, 2024-06-09:
The new Cumulus applet is included in v1.7.1

Thank you @zerbian !!

(and thanks always to @ph1-xyz for the icons/animation ideas - I might have to save some of these for future use! ;)

closed 2024-06-09 by djphazer