djphazer/O_C-Phazerville #71: Glitch in quadraturia mode (1.7 and 1.Z)

Burdienco, 2024-04-27:
Did anyone had some dlitches in quadraturia mode. In my case it just once in a few seconds jumps to some randome value for just a few frames in video terms, so much less than a second. To tes it I put ampletude of LFO to 0 so on the scope it shows plane horizontal line, and once in 5 or 6 seconds it jumps up and it is displayed on the screen as well. It would be clear if I could show video or picture, but I already installed v 1.69, it works fine.

djphazer, 2024-04-27:
Can you clarify what hardware you're using?

By the way, we don't talk about v1.Z - it's not safe there. Too many Zaps. 😬

djphazer, 2024-04-30:
I'm able to reproduce on Plum OCP X with v1.7. Seems like the "Output range" (attenuation) parameter is broken? I've neglected to test Quadraturia very much, so maybe I broke it recently. I'll have to dig deeper...

djphazer, 2024-05-07:
Bug squashed, I think (i'm very tired) - https://github.com/djphazer/O_C-Phazerville/commit/e0a4006fb53573080ab81a9f35ff905f45f3b203

I still don't quite understand it... What exactly happens when you access an array at index -1? "Undefined behavior"? How did it work before? Why did it break? Spooky stuff...

Anyway, the fix is pushed to the main `phazerville` branch, so a fresh Custom Build will include it. Let me know if it clears things up!

djphazer, 2024-06-09:
Fixed in v1.7.1

closed 2024-06-09 by djphazer