djphazer/O_C-Phazerville #75: Acid Curds chord progressions not saved

djphazer, 2024-05-19:
I've realized that since the `user_chords` array (in GlobalSettings) was replaced with `user_turing_machines` in Hemisphere, chord progression sequences made in the **Acid Curds** app are not properly saved in EEPROM. The app still works, and most other settings are saved, just not the selected chords.

This bug applies only to the Phazerville fork.

djphazer, 2024-05-31:
One potential solution to this that I'm considering is for **Acid Curds** to take precedence when enabled, and simply replace `user_turing_machines` with `user_chords` in that case - they both use the same block of 160 bytes. This seems like a good compromise for Custom Builds, but it will also apply to the Teensy 4.x builds with all the apps - **Enigma** won't be able to recall any of its TuringMachine's if **Acid Curds** is present. This issue will go away someday if I implement LittleFS or SD card storage for more space on Teensy 4.x.

Do more people care about saving Chord Progressions? or a library of Turing Machines?

An alternate solution is to store `user_chords` with the rest of Acid Curds' app data, but that will overflow EEPROM on Teensy 4.0. We just can't have it all, folks.

saamirr, 2026-03-15:
> One potential solution to this that I'm considering is for **Acid Curds** to take precedence when enabled, and simply replace `user_turing_machines` with `user_chords` in that case - they both use [...]

on Teensy 4.1, how does one access the Enigma sequences? The default build has a very limited number of the full screen apps, which makes sense since we'd want to run quadrants but I'm still working on using some of the features I'd ignored in my 3.2 ...

djphazer, 2026-03-15:
You would have to customize and build it from source to re-enable the Enigma app for T41... (and disable some other stuff to make room)