djphazer/O_C-Phazerville #83: MIDIOut seems to choke

xmacex, 2024-06-18:
Hi. You are doing a very good moving this classic, important module forward! Thank you for that.

Am I imagining things, but does the *MIDIout* app tend to choke, and stop sending stuff after a while? My untested hypothesis is that it's queue or something fills up, and freezes. Other things remaining constant, power cycling the module starts MIDI flowing again.

I was looking quick at MIDI monitor running on monome norns, but didn't have the possibility to record good data to share for the time being.

The MIDI notes are only sent if the note is different from the last one.

https://github.com/djphazer/O_C-Phazerville/blob/phazerville/software%2Fsrc%2Fapplets%2FhMIDIOut.h#L71

Would such filtering make sense for MIDI cc as well? More specifically if the MIDI cc has change, not the voltage which is much more sensitive. Would this have downsides for some MIDI use scenarios?

Best

djphazer, 2024-06-18:
It doesn't surprise me if there are bugs - I haven't fully tested and evaluated everything related to MIDI Output. Indeed, when the buffers fill up, bad things seem to happen! Needs more testing, trial & error.

Some things have been partially rewritten, and other things - like the MIDI-Out applet - are still mostly the original code from Hemisphere. I'm gradually working towards extracting all MIDI handling to the HSIOFrame or maybe even its own "driver" for simplicity. Reimplementing stuff as callback functions might be necessary.

If you come up with any improvements to the existing code, please share! Translating CV to MIDI messages is a more complex task than I initially thought... we want to avoid flooding the bandwidth with unnecessary redundant messages, which involves a lot of state and checks, but we also need to keep it efficient enough for the Teensy 3.2.

djphazer, 2024-06-20:
To directly answer your question about MIDI CC - yes, it would be desirable to cache it and only send it when it has changed!

xmacex, 2024-06-21:
I'm away from my o_C until sometime next week (waaah! 😭 ) to compile and test whether that helps the issue... and also consider about edge cases and potential scenarios where _not_ caching could be useful or interesting, but I have [a straightforward sketch](https://github.com/xmacex/O_C-Phazerville/tree/midi-out-cc-cache) for myself to check out. I'll send PR if it seems to help.

Might look at parametrizing the CC to another number than 1 (mod wheel) while I am at it.

closed 2025-01-04 by xmacex