djphazer/O_C-Phazerville #84: Calibr8 wider scaling factor

synthi, 2024-06-26:
Trying Calibr8 for tracking individual osc inputs in the Buchla 208 (the freq and rate inputs are more like 2V/oct instead 1.2V/oct). I see Calibr8 scale "only" goes to 105%, of course enought for normal duties, but having up to 200% could open potentially tracking  a lot of non standard vcos!

djphazer, 2024-06-26:
If you're looking for that kind of wide-range scaling, probably try AttenOff, which does (+/-)200%. Calibr8 is just for fine-tuning really...

I might try to add a config option somewhere for switching to different DAC voltage scalings.

synthi, 2024-06-30:
just a questions about calibr8 and calibr8r, does the calibrtion persist after changing the applet ? I mean, the calibrated profile will be available for the other applets?
ITOH tried AttenOff and worked great! I needed to go 140% scaling :)

djphazer, 2024-06-30:
Tracking correction from Calibr8 or Calibr8or does not affect other apps or applets... yet 😉

synthi, 2024-06-30:
yet!! :)