djphazer/O_C-Phazerville #88: ADSREG decay stage skipped when sustain is lowest

charonme, 2024-07-18:
when sustain is the lowest only the attack part is generated - probably caused by [this line](https://github.com/djphazer/O_C-Phazerville/blob/ec75211452d1a1126b69c6103ad9d9fee683f25a/software/src/applets/ADSREG.h#L312) in DecayAmplitude() that was there since [hemisphere](https://github.com/Chysn/O_C-HemisphereSuite/blame/893deeb27ecacddad638ff9180f244a411623e35/software/o_c_REV/HEM_ADSREG.ino#L227)

djphazer, 2024-07-19:
Good call! It looks like that line was supposed to be `if (sustain == 255)` so it will only skip the decay stage if sustain is maxed.

Just tested it out, seems good. Fix pushed - https://github.com/djphazer/O_C-Phazerville/commit/db2c9e10a8ecbfad7e90150bcf6074d695bb99df

djphazer, 2024-08-06:
Fixed in v1.8

closed 2024-08-06 by djphazer