djphazer/O_C-Phazerville #91: Spi add on for big screen oled

themangoest, 2024-07-22:
Hello,
Is it possible to have the spi connection add on for big screen oled? 
Eh2k from squares and circles did a wonderful job there. They are added on ports 26 and 27.

themangoest, 2024-07-22:
[image]

themangoest, 2024-09-03:
Is it possible to add this?

djphazer, 2024-09-03:
This is a bit outside my wheelhouse, although I am interested... more details?

themangoest, 2024-09-03:
Hi, thanks for checking this out.

the "Big SSD1306" the SPI port is operating at 8Mhz, else the display stay black.

you could connect the display to the second SPI Port (SPI1 - Pin 26;27)

the DAC connection is the same, expect not sharing 11 and 13 with the display. The alternative spi1 display connection is 26 and 27. 

In squares and circles you an option for that in the web programmer. See picture above.

Banshee700, 2024-09-18:
Yes, we also would like to use the big OLED for the upcoming 4UBu version of it.

djphazer, 2024-10-28:
For reference, this is the code fork with the driver tweaks to make this work - https://github.com/mcleinn/O_C-for-SSD1309-2.4-inch-display (I was previously unaware this existed!)

I've cherry-picked/merged the changes on a separate branch (dev/large-OLED), and verified that it runs on Teensy 3.2, but of course the standard small display just looks garbled. If someone can verify that it works, I'll try to get it working for T4.x as well.

Experimental builds for testing: https://github.com/djphazer/O_C-Phazerville/actions/runs/11538938594

Banshee700, 2024-10-29:
Awesome :)
I just ordered two of the OLEDs myself.
I think the main target is the T4.1 as that is the one used in Pauls new
design, right ?
Maybe i should start to breadboard something like that...
Just to be flexible with pin assignments and stuff.

best
Sascha

themangoest, 2024-10-29:
Hi Nicholas,I tested with 2 different teensy 3.2 and get the same screen issueMaybe it is better to concentrate on the teensy 4 since the 3.2 is obsolete?Cheers Gregory Verstuurd vanaf mijn iPhone

djphazer, 2024-10-30:
@themangoest - You tested with Teensy 3.2 using the 2.4" OLED and it didn't work? I'm just trying to confirm a known-good scenario with this existing driver code modification. All the driver code for Teensy 4.0 and 4.1 is a bit different, but if I know what works, maybe I can figure out what to copy-paste into where...

Perhaps I should acquire a breadboard and various parts myself and actually learn how to swim!

themangoest, 2024-10-30:
Hi Nicholas,You want me to send a pcb and screen?Cheers Gregory Verstuurd vanaf mijn iPhone

djphazer, 2024-10-30:
I've ordered a 2.4" OLED, as well as a breadboard kit. I have the original 14HP O_C hardware to try it on, as well as the new 4.1-based unit. End of this week or next, I'll be ready for prototyping!

Banshee700, 2024-10-31:
Perfect :)
I's it easy to access the pins on Paul's beta unit ?
Do you plan to cut and wire or can one solder on the top pins of the T4.1 ?
As in making a cable loom that attaches to that ... that would be cool
Mine is on the way as well...

djphazer, 2024-11-04:
[image]

Some success with Teensy 4.1 on original hardware! Simply setting SPI clock to 8Mhz works, but I had to reduce the core ISR frequency to get it stable... perhaps with a bit more determination, I can get the subpage transfer working properly.

For reference, it's this item on Amazon:
ACEIRMC 2.42 inch 128x64 OLED LCD Display Module SSD1309 7 Pin SPI/IIC I2C Serial Interface for Arduino UNO R3 (White) https://a.co/d/hFrPwds

Banshee700, 2024-11-05:
My old eyes love to see that :)
Thanks for the link...
Thats is like less than half what our Amazon charges us in the EU
Also it has the pins in the center which is cool...

themangoest, 2024-11-05:
Hi all,Yes it’s the same oled that I use but I have the pins on the side.Looking forward to the next chapter of progress!Keep it up 👍 Gregory Verstuurd vanaf mijn iPhone

djphazer, 2024-11-05:
Success! Subpage transfer is working, with full speed ISR. Works on T3.2 and T4.x. It also still works with the smaller screens, so win-win!

It will work with the original display refresh rate (1ms redraw timeout) but I dialed it in at 2ms to eliminate some visual glitches. This means things like the scope_render screensavers operate somewhat slower than before.

Latest builds for testing: https://github.com/djphazer/O_C-Phazerville/actions/runs/11681824832

themangoest, 2024-11-05:
Hi Nicholas,

Success here too with the Teensy 3.2
I'll try the T4 in 6 hours and will keep you posted.

you rock!

cheers

*Gregory Delabelle*
MANGOEST
Ⓣ +32 485 985644
@ www.mangoest.be
<https://github.com/themangoest/>

themangoest, 2024-11-10:
Hi Nicholas I guess the next logical step is trying the new hardware OC with more in and outs?Dobyou have the schematics for that?Cheers GVerstuurd vanaf mijn iPhone

djphazer, 2024-11-10:
New hardware schematics here: https://github.com/PaulStoffregen/O_C_T41/tree/main/schematic

It should work no problem, since the new hardware has Display and DAC on separate SPI buses.
I still need to try it on one of my beta units, but I'll have to desolder the other screen... probably make a cable loom like Sascha suggested.

Banshee700, 2024-11-11:
I just tested the new OLED with a hOC and yes, that is a plug in
replacement and works fine with the new firmware :)
[image: 9a724fde-72da-4275-b17b-daed0eb54817.jpg]
Cable loom was the solution indeed, but our displays are in sockets, so i
just had to pop it out, add some 90 degree pin headers and voila..

Regarding the new one...i wonder if we should just make some GErbers from
the KiCad files or start to build the 4U version right away.
I assume Gregory would like to make a Serge version, and I have a concept
for the Buchla one that looks like this :

[image: 2OCx.1.1.jpg]

This time the USB connections need to be on the front, and the attenuation
for both input and output still is very important i think.

djphazer, 2024-11-11:
oof, can't see the photos when you comment via email reply... happy to hear of some success with the hOC!

Banshee700, 2024-11-11:
[image]

> I just tested the new OLED with a hOC and yes, that is a plug in replacement and works fine with the new firmware :) Cable loom was the solution indeed, but our displays are in sockets, so i just had to pop it out, add some 90 degree pin headers and voila.. Regarding the new one...i wonder if we should just make some GErbers from the KiCad files or start to build the 4U version right away. I assume Gregory would like to make a Serge version, and I have a concept for the Buchla one ready This time the USB connections need to be on the front, and the attenuation for both input and output still is very important i think.

djphazer, 2024-12-18:
Large OLED support has been merged in for v1.8.3! Seems pretty stable on all the platforms, let me know if you run into any issues.

themangoest, 2024-12-18:
Dope,I have to work on the next gen oc :)Is that firmware already released too?Cheers GVerstuurd vanaf mijn iPhone

djphazer, 2024-12-18:
Yes, indeed! PSv1.8.3 is published with it now.

themangoest, 2025-05-17:
Hi Nicholas,Hope all is ok.I have some builders who can use firmware 1.9 with the big oled.Has something changed in the firmware by any chance?If you have to time can you send me the latest hex for 4.0 and 4.1 for big oled support ?Cheers Gregory Sent from my iPhone