djphazer/O_C-Phazerville #93: Tuner in v1.8  on T40 crashes in flipp_180

TFModular, 2024-08-07:
When I select the tuner on the left-hand side, the module suddenly jumps into debug mode and I can't get out of it. Only with a restart. This obviously only happens in the FLIP_180 installation.

djphazer, 2024-08-07:
Roger that, thanks for reporting! The T40 flipped build could use more testing on my end...

djphazer, 2024-08-10:
Just pushed a fix for this - https://github.com/djphazer/O_C-Phazerville/commit/7014a5691af665b17f296dfd828ee1c4cfcf2523

Grab the artifact from the latest workflow run to test - https://github.com/djphazer/O_C-Phazerville/actions/workflows/firmware.yml?query=branch%3Aphazerville
...or wait for v1.8.1 ;)

djphazer, 2024-09-12:
This is stable in v1.8.1

closed 2024-09-12 by djphazer