djphazer/O_C-Phazerville #94: Sequence X dual mode proposal

pyerbass, 2024-08-08:
Hi! what if, when both applets are set on sequenceX, the app becomes a single 16 step sequencer, still with two outputs and two clock inputs, like two playheads moving along the same sequence at their own pace.

Gate in A: CH1 clk
Gate in B: CH2 reset
Gate in C: CH2 clk
Gate in D: CH2 reset

CV in A: transpose steps (or transpose CH1)
CV in B: randomize steps
CV in C: transpose selected step (or transpose CH2)
CV in D: randomize selected step

OUT A: CH1 v/oct out
OUT B: CH1 step1
OUT C: CH2 v/oct out
OUT D: CH2 step1

zerbian, 2024-10-06:
Sounds like a interesting idea. I already implemented some Hemisphere applets, but I haven't yet looked into syncing two applets, some maybe @djphazer can give me give hint where to start. I love the idea and would start working on it.

djphazer, 2024-10-06:
You could look at the **ASR** applet and its `HSRingBufferManager.h` for an example of how this is currently achieved. Other examples are the `HSLorenzGeneratorManager.h` used by LowerRenz and the `HSProbLoopLinker.h` used by several different trigger/melody generator applets.

I'm discouraged from propagating the pattern of singleton classes because it feels like an unnecessarily complicated abstraction. Perhaps it's beneficial for threaded server applications or something, but in our firmware context, two applets are not actually separate threads; they are simply processed in sequence - left side first, then right side, once every _tick_. To "link" them together is merely a matter of shared global state.

In fact, I've already converted the HS::RingBufferManager to a simpler, static struct object... but, it still seems buggy. (The DMAMEM allocation makes it start out with random data; maybe that's why)

Anyway, @zerbian hopefully that gives you some insight! Maybe there are simpler solutions out there; maybe not. Perhaps it could be helpful if an applet could directly talk to its neighbor without a mediator. ;)

zerbian, 2024-10-06:
Thank you for your insight and starting point. I will look into these applets!

zerbian, 2024-10-13:
@pyerbass How would you imagine the navigation and controls will work? I think the easiest solution will be: each half only controls its 8 steps, and only the playheads "cross" the different applets.