djphazer/O_C-Phazerville #95: Question about DualTM and locking mechanism

gilbertohasnofb, 2024-08-11:
Hi, I've been using the DualTM but I have a question about its functionality and whether I'm perhaps misunderstanding its use or not. 

The DualTM shift registers seem to run from right to left, and I believe that a flipped bit will be the rightmost one. The original Turing Machine by Music Thing Modular runs left to right, with the leftmost bit being the flipped one. This difference is important because the pitch output in both cases will be the leftmost n bits, set by the length value.

A big consequence of this is that when we lock the original Turing Machine (i.e. equivalent as p=0 in DualTM), we lock the pattern of notes that we literally just heard. In the case of DualTM, we are NOT locking what we just immediately heard, but what would have been heard in the future, as those bits are still not within the n-range of whatever length you haver set it.

This should be easily reproducible in practice: set the length of your pattern to something short and memorizable like 6, then set p to 50% so that the pitch output becomes completely random. As soon as you hear something memorable, lock p to 0% and notice how that pattern is not captured but rather something else, which then of course gets repeated over and over. The reason is you locked whatever content was in bits 26 through 32, but what you just heard was on bits 1 through 6. Compare this with the behaviour of the original Turing Machine seen here at 1'23": https://www.youtube.com/watch?v=-Q6dXbtNh0Q&t=83s

I think that simply changing the direction and flipping the first bit like the original Turing Machine would solve this (unless I'm completely misunderstanding the behaviour of DualTM).

djphazer, 2024-08-11:
I know, there's a subtle but important difference in behavior, and it bugged me for a while ... but you're in luck - you can now switch the shift direction in v1.8! Use the AuxButton action on the Length parameter to flip it.

gilbertohasnofb, 2024-08-11:
That's really excellent, I'm super happy about this! I will try the new release shortly.

May I ask if this behaviour is now the default one? In case not, it might be worth considering to make it the default so that the original Turing Machine behaviour is emulated.

Thank you so much for this firmware and for the constant updates, your work is truly wonderful.

djphazer, 2024-08-11:
Good call, I'll definitely switch the default! ➡️

djphazer, 2024-09-12:
Right-shift behavior is now the default in v1.8.1

Cheers!

closed 2024-09-12 by djphazer