djphazer/O_C-Phazerville #97: MIDIIN: saving preset clears VOR setting

jameswiegand, 2024-08-19:
When I use the save O_C (long press on left encoder) the VOR setting returns to default. The O_C is running MIDIIN on both sides.

Build o_C-phazerville-v1.8_phz+VOR-ed61862c on Plum Audio OCP. Not sure if this is intended behavior?

-james

djphazer, 2024-08-19:
Thanks for pointing this out! The Vbias setting is not currently stored for Hemisphere, so it will probably revert back to ±5v anytime you exit to the App menu and then return...

There is room now to store the setting, I just hadn't thought about it. I'll try to make it persistent in the next update!

Would you prefer a separate Vbias setting in each Preset? Or one global Vbias setting for Hemisphere?

jameswiegand, 2024-08-20:
Hi,

Wow thanks for the quick update! I would use different "Vbias" depending on
the function, so yeah saving with each preset would be awesome.

Thanks,
-james

djphazer, 2024-09-12:
The Vbias setting is now saved per-preset in v1.8.1

Cheers!

closed 2024-09-12 by djphazer