djphazer/O_C-Phazerville #103: Cannot calibrate in 1.8.1

cdromain, 2024-09-27:
Hey Nick,

Hope this message finds you well. First huge kudos & thanks for your amazing work on the O&C development, Phazerville is plain awesome 🙌🏼 I had to buy you a nice Ko-fi 😃

Just updated to FW 1.8.1 using a custom build (`/buildthis Quantermain, Piqued, Quadraturia, Acid Curds, Passencore, LORENZ, References`) and noticed that the Harrington 1200 app wasn't tracking properly the 1v/o pitch CVs sent by my Shakmat Bard Quartet quantizer (like 1 semitone off). So I tried to calibrate the ADCs in the Setup/About app as shown in your [YouTube short](https://www.youtube.com/shorts/AIadpDclP7M) but in the calibration page, when I click on OK the bar at the bottom of the screen fills up without any popup screen showing up, there's nothing I can do to actually run the calibration routine. Also when changing the Use Defaults value the module ends up freezing on certain values.

In the 4/4 ADC page I noticed what looks like high values for C1-C4 (in the 20-30s).

I'm running Phazerville 1.8.1 on a Teensy 4.2 (Big T Music build).

Thanks in advance for your support & keep it up, cheers from France !

Romain

djphazer, 2024-09-27:
Have I broken something on T3.2? I'll investigate...
Could you maybe shoot a quick video of the problem?

cdromain, 2024-09-27:
Thanks man 🙌🏼

Here are 2 rough pics to illustrate the issue (nothing to shoot really) : 

1. You can see that each time I press [OK] (right encoder) the bar fills up but nothing else shows up on screen, nothing I can do to actually run the calibration :
[image]

(And the O&C freezes when reaching certain "Use Defaults ?" values)

2. The ADC values : (drifting slightly)
[image]

Hope that nothing's wrong with my O&C - thanks again !

djphazer, 2024-09-28:
Definitely a firmware bug... I'm able to reproduce with your specific custom build, but it works fine in the v1.8.1 release build. Strange, but a good clue... seems like it's broken without the `PEWPEWPEW` flag.

While I work on a fix, as a workaround, you could flash to the release build to calibrate, and then flash back. Calibration data is shared and will persist between different builds.

djphazer, 2024-09-28:
'Twas a small oversight, easy fix - https://github.com/djphazer/O_C-Phazerville/commit/c41cdb439c3efdf960cb173852fe926a2a5c405c
If you resubmit for a Custom Build, it should work!

cdromain, 2024-09-28:
Hey Nick,

Thanks so much for the quick fix - legend 🙌🏼

I confirm I could access the calibration routine properly and successfully removed those 20 something offsets (not sure why they were there in the first place) and have now ADC readings drifting around 0, so the quantization should behave much more nicely.

Cheers and beers on me if we get to meet one day (SuperBooth maybe ?) 🍻

Romain

closed 2024-09-28 by cdromain