djphazer/O_C-Phazerville #104: Screensaver doesn't start on o_c t3.2

nvmker, 2024-10-03:
Hi,
I noticed that the screensaver doesn't start on the o_c teensy 3.2 as in this constants:
```
static constexpr uint32_t SCREENSAVER_TIMEOUT_S = 25; // default time out menu (in s)
static constexpr uint32_t SCREENSAVER_TIMEOUT_MAX_S = 120;
```
I switched from hemisphere, was working well there.

djphazer, 2024-10-03:
It works for me! Note that it's actually in minutes, not seconds, but that was true in Hemisphere as well:
https://github.com/djphazer/O_C-Phazerville/blob/phazerville/software/src/OC_ui.cpp#L184-L186
Maybe I'll refactor the variable names to reflect this.

The runtime setting is stored in calibration, if you want to change it.

nvmker, 2024-10-04:
@djphazer I just tested it, waited more time and it works, thank you.
Is there a way to change timeout without recalibrate the module? I had a bad history with this cheap oled screens burnout with different modules

djphazer, 2024-10-04:
Running thru the Calibration shouldn't hurt anything, just use the left encoder to skip thru to the end... probably set screensaver to 1 minute for conservation. And then also select the "blank" screensaver in [Hemisphere Config](https://firmware.phazerville.com/Hemisphere-Config). "Zips" is also designed to prevent burn-in.

nvmker, 2024-10-05:
> Running thru the Calibration shouldn't hurt anything, just use the left encoder to skip thru to the end... probably set screensaver to 1 minute for conservation. And then also select the "blank" [...]

Thank you! I will try.

closed 2024-10-05 by nvmker