djphazer/O_C-Phazerville #109: External Clock no longer works in the Hemisphere Suite in v1.8.2.

TFModular, 2024-11-24:
Seems to affect all apps in Hemisphere. The original apps like Sequins work, only Hemisphere does not. Am I missing something?

djphazer, 2024-11-24:
None of the trigger inputs work in Hemisphere? Can you provide some more details? Perhaps a screenshot of your [Clock Setup](https://firmware.phazerville.com/Clock-Setup) screen settings?
Did this work fine in an older release of Phazerville? Are you running a Custom Build for T3.2? Or is it T4.0?

TFModular, 2024-11-24:
It is T4.0. v1.8.1 works without any problems. The firmware is freshly installed. The settings in the clock setup have not been changed.

djphazer, 2024-11-24:
Very strange... not much changed between v1.8.1 and v1.8.2. I just reflashed my T4.0 unit back to the release build (085b225e) and could not reproduce - applets respond to all 4 trigger inputs, and I can see them visually on the Meters screensaver. I even did a Reset to clear the EEPROM settings... of course, calibration is still retained - I'll admit, it's hard for me to test for bugs that may happen with a completely fresh Teensy, uncalibrated.

Another thought: all the internal clock multipliers are OFF by default now, which may give the impression that the internal clock doesn't work. This wouldn't affect the trigger inputs, though.

TFModular, 2024-11-25:
Yes, really strange, I can no longer reproduce the error either ... I may have triggered a bug without knowing how. I will continue to monitor this and report back if there are any findings. Thanks for the quick response.

djphazer, 2024-11-26:
Another theory: This could also be related to the new Tuner code not shutting down properly somewhere... which I think would disable TR1.

closed 2025-01-28 by TFModular