djphazer/O_C-Phazerville #112: THIS IS MY OPINION NOT A BUG: v1.8.3 ClockSetup SLIDEOUT_TIME is much better at 50 ms.

beau-seidon, 2024-12-27:
not sure this if warrants an issue, but initially i found the new ClockSetup interface to be irritating. 
i determined that my entire fuss was caused by the sloow slide down from the top.
i decreased the SLIDEOUT_TIME (ClockSetup.h, line:27) by half and it still showcases the nifty animation, but its better and now the split style interface no longer bothers me.

djphazer, 2024-12-27:
I'm inclined to agree - it should be snappy and non-intrusive. I implemented it quite recently, so it didn't get extensive tuning before release. This nitpick highlights some important observations about timing...

The catch is: that value is not in _milliseconds_. Since it decrements the `slide_anim` counter in the View thread, the value is in _frames_, and that depends on the refresh rate... which I also recently adjusted. Original stock firmware had `REDRAW_TIMEOUT_MS=1` for blazingly fast refresh rate, but I've bumped it up to 4ms to relieve the CPU to compensate for various things. Theoretically, assuming it takes 1ms to draw the screen and loop back around, that's about 200Hz refresh rate.

The View thread happens in the main loop() and has lower priority than the ISR, so it will actually run slower when there is higher load in the Controller thread - you can see this visibly if you load Ebb&LFO on both sides.

We can get more accurate and consistent results by simply moving the counter decrement to the Controller instead. The ISR fires every 60us - 16666 ticks per second.

djphazer, 2025-01-11:
I addressed this in https://github.com/djphazer/O_C-Phazerville/commit/376fac357cb42b44953d76970eae676b09f4a017

Feel free to tweak that timing value to your liking.

closed 2025-01-11 by djphazer