djphazer/O_C-Phazerville #114: New calibration screen issues

Terumi, 2024-12-30:
Hello, I cannot seem to calibrate to 6V.
I have two micro o_c and the maximum value I can output (where the value of the right encoder is maxed out to something like 6555....) is 5.94V. This messes up the whole calibration procedure.
Also, could It be (with a button press or something) to have corse steps on the encoders when calibrating?

djphazer, 2024-12-30:
Oof, I'll have to revisit this then. Thanks for reporting. Maybe we could use the 5V point for max instead...
Meanwhile, use an older version for calibration!

Terumi, 2025-01-02:
I though that calibration was lost when reflashing! Thanks, will try it!
Happy new year mate!

closed 2025-01-02 by Terumi

djphazer, 2025-01-02:
No worries! The calibration data is still backwards compatible with stock firmware, always retained when reflashing.

I have a fix to try tho, using -3v and 5v. When I confirm that it's good, I'll push to the main branch and it will be available for Custom Builds.

Terumi, 2025-01-02:
I just flashed a new, custom firmware...
The calibration got lost :/

It would be great if you could have a combination of both...
Set it up with -3V and 5V, and then, in a fine calibration screen you could do adjustments if needed!