djphazer/O_C-Phazerville #115: section `.text' will not fit in region `FLASH'

xmacex, 2025-01-04:
Wupsi when i `pio run -e pewpewpew`, i get

```
Indexing .pio/build/pewpewpew/libFrameworkArduino.a                         Linking .pio/build/pewpewpew/o_C-phazerville-v1.9pre_phz-4577d6b7.elf
/home/we/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: .pio/build/pewpewpew/o_C-phazerville-v1.9pre_phz-4577d6b7.elf section `.text' will not fit in region `FLASH'
/home/we/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 31660 bytes                                                        collect2: error: ld returned 1 exit status
*** [.pio/build/pewpewpew/o_C-phazerville-v1.9pre_phz-4577d6b7.elf] Error 1 ======================= [FAILED] Took 191.16 seconds =======================
                                                                            Environment    Status    Duration
-------------  --------  ------------                                       pewpewpew      FAILED    00:03:11.158
================== 1 failed, 0 succeeded in 00:03:11.158 ==================
```

I'm on

```sh
⠃▋⡇ ~/software/O_C-Phazerville/software$ git log -n1
commit 4577d6b7086a89fe40ab7f6a8c5c64b4dc3998b1 (HEAD, origin/dev/1.9)      Author: Nicholas J. Michalek <djphazer@gmail.com>
Date:   Fri Jan 3 22:25:22 2025 -0500                                       
    VectorLFO: Fix sync freq display; move some functions                   ⠃▋⡇ ~/software/O_C-Phazerville/software$
```

Compilation worked fine two weeks ago before xmas.

djphazer, 2025-01-04:
Apologies - I've moved over to using "prod" as the main build target. Obviously the "pewpewpew" config has too many things now... comment out one or two apps if you must to make it build again.

Thanks for testing out the bleeding edge v1.9 beta stuff!

xmacex, 2025-01-05:
Ah yes got it, and it compiled. I was on a habit and used my command history plus what the README said/says.

Fun! Viznut cracker, sweet screaming relaxing lullabies here. My fantasy is someday maybe succeed in bringing it over from applets to apps.

Anyway thank you very much for this firmware, it's really incredible job ⚡⚡⚡

closed 2025-01-05 by xmacex